OctoPrint/OctoPrint #1837: Printer command timeout appear to be corrupting internal print metadata

fraschetti, 2017-03-25:
#### What were you doing?

My printer had a new firmware available which caused me to acknowledge physically at the printer each time I started a print via Octoprint.
If I didn't physically press the button on my printer, a timeout error occurs with OctoPrint, the print is marked as failed but the print information (which is normally otherwise available on the failed event) is unavailable.

#### What did you expect to happen and what happened instead?

It should be business as usual interms of the PrintFailed event having full knowledge of the print that failed. The exception (see below) is likely the cause.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

OctoPi 0.13

#### Printer model & used firmware incl. version

(after I applied my firmware update)
Original Prusa i3 MK2
Marlin V1.0.2; Sprinter/grbl mashup for gen6

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

Apologies but I only copied the relevant log entries and have since updated my firmware so I cannot reproduce.

https://gist.github.com/fraschetti/77012a8b39640bad0cac0a78e25d0c50

#### Link to contents of terminal tab or serial.log

I didn't capture this information before updating my firmware (which makes the issue unreproducable)

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) or video(s) showing the problem:

N/A

foosel, 2017-03-27:
It's not corrupting it, but it's trying to access data that is no longer available because it's been reset from a different thread.

I should have solved that now through the above commit, currently on the `maintenance` branch, soon on the `devel` branch as well and in any case to be released with 1.3.3.

foosel, 2017-05-31:
1.3.3 was just released.