OctoPrint/OctoPrint #1839: Use folders on local NAS

Gandohr, 2017-03-26:
I build a raspberry pi 3B with the standard octopi image.
Then i created a folder called nas in the home of pi.

in fstab i made a //nasaddress/folder /home/pi/nas gifs username=username,password=password 0 0 line :)
That seems to be working and nas is mounted at boot.

in octoprint i set the upload, time-lapse and watch folder to
/home/pi/nas/uploads, and of cause /timelapse and /watch.
When i upload a stl file in octoprint it gives an error
looking at the error its an internal server error 500.
More investigation gave me http://<localip>/api/files/local/ and some thing about not having an API key.

Strange thing is the file is uploaded and is on the correct folder on the nas, after a reload the file is visible in octoprint and i can even slice it. al rights on all folders are drwxrwxrwx at this moment.

Its not that i wanna share my files across multiple servers, its just my nas has more disc space then my raspberry. The folders are only being used by that one instance of octoprint, and only trough octoprint (no files will be added to the upload folder in any other way.

suggestions on how to use some external device as my storage?

Thanks

Gand

foosel, 2017-03-27:
This sounds more like a support request than a bug report or feature request. The mailing list or the G+ community as linked above by the bot are a better place for that - way more people monitoring those than here.

Since you mentioned you received a server error, you should also include `octoprint.log` in your problem description.

closed 2017-03-27 by foosel

Gandohr, 2017-04-03:
I still think its a bug.
/home/pi/nas is connected to my NAS as a cifs share.
in octoprint i've set the upload folder to /home/pi/nas/Uploads

When i upload the file it is written in the right folder (so octoprint has write permissions)
But still gives an error.
When i look in the folder, the file and the .metadata.yaml are both there.

The error comes at the moment normally the cura slice popup would display (that does not happen,).
The file will show up in octoprint, en then i can slice it without any problems.
reason i did not provide a log, cause there is nothing in there.

I removed the log, restarted octoprint and did 2 uploads that gave an error.
this is the log
2016-03-18 08:17:15,523 - octoprint.server - INFO - Starting OctoPrint 1.2.10 (master branch)
2016-03-18 08:17:15,586 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-03-18 08:17:16,145 - octoprint.plugin.core - INFO - Found 5 plugin(s) providing 5 mixin implementations, 2 hook handlers
2016-03-18 08:17:16,398 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-03-18 08:17:16,400 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-03-18 08:17:19,191 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 8.1.1
2016-03-18 08:17:19,200 - octoprint.plugin.core - INFO - Initialized 5 plugin implementation(s)
2016-03-18 08:17:19,351 - octoprint.plugin.core - INFO - 5 plugin(s) registered with the system:
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/virtual_printer
2016-03-18 08:17:19,352 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-03-18 08:17:19,356 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-03-18 08:17:19,357 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-03-18 08:17:19,358 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-03-18 08:17:20,256 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-03-18 08:17:20,273 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2016-03-18 08:17:20,276 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2016-03-18 08:17:20,501 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2016-03-18 08:17:20,587 - octoprint.server - INFO - Listening on http://127.0.0.1:5000