OctoPrint/OctoPrint #1840: HTML Username and Password Not Used When Taking Timelapse Snapshot

Akalay27, 2017-03-27:
#### What were you doing?

I was setting up my webcam html address in the settings window, which included an embedded username and password: 
`http://admin:@10.0.1.19/image.jpg`
The test button works, and shows the correct image.

#### What did you expect to happen and what happened instead?
I expected to create a successful timelapse, but instead I was confronted with
 
Could not capture snapshots
Failed repeatedly to capture timelapse frame from webcam - is the snapshot URL configured correctly and the camera on?
Snapshot Url: `http://10.0.1.19/image.jpg`
Error: `401 Client Error: Authorization Required`

It seems to not be including the `admin:@` in the html request.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (Master Branch)

#### Operating System running OctoPrint

Raspberry Pi running OctoPi Jessie

#### Printer model & used firmware incl. version

Wanhao Duplicator i3

#### Browser and Version of Browser, Operating System running Browser

Chrome, Macosx

I have read the FAQ.

foosel, 2017-03-27:
Are you sure you clicked "Save" after you set the snapshot URL with included credentials?

I cannot reproduce the problem over hear, entered credentials into the URL, saved, timelapse captures are now done correctly with included credentials. Example (debug logging enabled):

```
2017-03-27 08:45:08,206 - octoprint.events - DEBUG - Firing event: CaptureStart (Payload: {'file': '/home/pi/.octoprint/timelapse/tmp/20mm-box_20170327084348-8.jpg'})
2017-03-27 08:45:08,271 - octoprint.timelapse - DEBUG - Image /home/pi/.octoprint/timelapse/tmp/20mm-box_20170327084348-8.jpg captured from http://foo:bar@127.0.0.1:8080/?action=snapshot
2017-03-27 08:45:08,273 - octoprint.events - DEBUG - Firing event: CaptureDone (Payload: {'file': '/home/pi/.octoprint/timelapse/tmp/20mm-box_20170327084348-8.jpg'})
```

Akalay27, 2017-03-27:
Sorry, I guess I never actually pressed save when exiting the menu.

closed 2017-03-27 by Akalay27

foosel, 2017-03-28:
Easy mistake! :)