OctoPrint/OctoPrint #1842: [Brainstorming] FontAwesome in OctoPrint is really outdated

Depaulicious, 2017-03-27:
As a plugin developer I sometimes struggle to add the right icon for each feature. Very often all icons I need are in latest FontAwesome, but not in the one included in OctoPrint.

I'm opening this issue to learn if you guys are okay with an eventual pull request in which I'll try to update it , trying not to break everything else (e.g. making sure the icons that are currently provided are also available as `icon-*` in addition to `fa-*`, or other similar solutions).

I'm not willing to work on a fork that won't be accepted upstream; plugin compatibility comes first IMO.

foosel, 2017-03-27:
If it's possible to stay backwards compatible (icon-* works, existing plugins don't break), that would be **great**. The backwards compatibility situation is actually the reason why this stuff is so outdated (well, that and the fact that I didn't think about doing an update before rolling out plugin support with 1.2.0). The last time I looked it sounded like being quite a huge ordeal though and hence I haven't yet found the time. Any help there would be very welcome indeed!

I'd target that for 1.4.0 rather than 1.3.x though - that way we'd get more time to iron out any compatibility kinks before it hits mainstream, and allow to collect feedback from people running the  `devel` branch.

Depaulicious, 2017-03-27:
Awesome! I'll try to do it as soon as I have time, then.
I think, however, that the icons changed their look a little bit with releases. I'll see how they look like.

foosel, 2017-03-27:
Definitely keep me updated! If there are some changes, that's a death we'll have to die I guess, and it will not cause as much interruption as a completely missing icon ;)

ntoff, 2017-03-27:
wait, if I'm using fa-* now, and it works, what's up with that? Does the current octoprint version of font awesome also have fa- fonts in it? Or have I somehow got a client side thing going on meaning only I can see the icons I'm using?

foosel, 2017-03-27:
@ntoff might be a plugin that concurrently installed a newer version of font awesome.

Come to think of it, that might actually be a good solution too (if it doesn't cause conflicts), marking old icons as deprecated but keeping them available for a couple more versions to give plugin authors time to migrate.

ntoff, 2017-03-27:
ruh roh, there's uh... some place I need to be, I think I see my friend over there

*runs*

foosel, 2017-03-28:
Well, at least that mystery is solved, the TouchUI plugin bundles a newer Font Awesome (4.4.0).

Which means that apparently it's indeed no problem to run a newer version concurrently with the old one. So now I wonder if it wouldn't be the best approach to just add the newer one, switch to that in core OctoPrint in the short run (1.3.3 or 1.3.4) and for 1.4.0 target a full replacement using compatibility classes on top.

I also briefly thought about template/assets preprocessing to simply replace css classes accordingly, but that might quickly backfire horribly.

In any case, the migration guide makes it appear to be way tamer than I feared: https://github.com/FortAwesome/Font-Awesome/wiki/Upgrading-from-3.2.1-to-4

This could probably be automated fairly easy for plugin authors too.

@BillyBlaze Does TouchUI use a customized version there or is it just vanilla? Asking because I'm wondering if bundling that with stock OctoPrint would cause problems in any way or not.

BillyBlaze, 2017-03-28:
Vanilla. I don't see any problems running this next to 3.2.1 and phasing it out.

Depaulicious, 2017-04-06:
Sorry for not following up but I've been busy last week, and will be busy for probably one or two weeks, during which I'll also have to fix my printer :'(

I'll try to do it ASAP.

foosel, 2017-04-07:
@Depaulicious no harm done! But please continue keep us updated :) It's not a problem if you can't find the time or the energy after all in a couple of weeks, it should just be communicated so that I know that someone else could/should take care of it :)

Depaulicious, 2017-05-13:
I almost forgot about this haha
I'll do it now

Depaulicious, 2017-05-13:
There it is, #1915 . I hope this doesn't cause issues with other PRs as there are a lot of changes in the templates (due to renaming `icon-` to `fa-`)

closed 2017-05-18 by Depaulicious