OctoPrint/OctoPrint #1843: self._printer.select_file(path, False) in plugin not working with absolute paths

tchapi, 2017-03-28:
#### What were you doing?

I have a very simple plugin that creates a gcode file on disk, in the plugin data folder.

    path = os.path.join(self.get_plugin_data_folder(), str(uuid.uuid4()) + ".gcode")

After this line, the file exists at `/home/pi/.octoprint/data/my_plugin/4baac482-8629-40ad-96ec-fcf23d64a05d.gcode`.
When I select the file for printing via `self._printer.select_file(path, False, True)`, Octoprint changes the path of the file and cannot find it anymore: 

    OSError: [Errno 2] No such file or directory: '/home/pi/.octoprint/uploads/home/pi/octoprint/data/my_plugin/4baac482-8629-40ad-96ec-fcf23d64a05d.gcode'

It seems to concatenate different paths instead of using my absolute path given to `select_file`.

#### What did you expect to happen and what happened instead?

I expect the file to be printed, and instead, Octoprint logs an error saying it can't find the file (with an incorrect path)

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2.post0.dev0+g0a69dbe (master branch)

#### Operating System running OctoPrint

Raspbian - not using OctoPi

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

https://pastebin.com/K53RiiRc

#### Link to contents of terminal tab or serial.log

N/A

I have read the FAQ.

Thanks a lot !
Cyril

foosel, 2017-03-29:
That is intentional behaviour. You are only allowed to select files from the uploads folder for printing, not arbitrary files from disk. You'll need to add the file to a storage first (`local` is the only one that makes sense right now), otherwise you won't be able to handle it. This is also to prevent you from trying to print `/etc/shadow` or `/tmp/insanely_huge_file.iso`. A file uploaded through the file manager will also have some basic sanitization done to it, plus trigger GCODE analysis and various other metadata related tasks.

If you want to create a GCODE file from your plugin on the fly, create it as a temporary file, then use `self._file_manager` to upload it, then trigger the print using the new path returned by that. See [the implementation of the watched folder handling](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/util/watchdog.py#L33-L71) for an example on how that would work.

tchapi, 2017-03-29:
Alright, that's understood, thanks a lot. I would maybe then add that recommandation in the docs, because I didn't find anything on the subject and that's why I went with the basic approach of writing the file to disk and selecting it.

Thanks for your detailed answer !
BR

closed 2017-03-29 by tchapi

foosel, 2017-03-29:
I just realized too that it falls on its face at a different location than I thought it would - only after actually selecting the file but trying to produce necessary job data for it. I'll have to adjust the handling so it does the check if it's on a registered storage earlier in the process, and while at that I'll also look into some better documentation (and error messages) regarding that.

foosel, 2017-03-29:
Done, see the commit above. Will be in 1.3.3, already on `maintenance` and soon `devel`.