OctoPrint/OctoPrint #1844: [Request] Set CPU Priority for Timelapse creation

fast930, 2017-03-29:
#### What were you doing?
Having to wait to start a new print while the timelapse is being created from my previous print.  This is to ensure that avconv doesn't contend with octoprint for CPU resource which may affect my print.

#### What did you expect to happen and what happened instead?
Wish I could print immediately.  If avconv were started with a CPI priority (nice value) lower than octoprint, the OS would give priority to octoprint and reduce the potential for contention.  Instead, I am urged to wait, often many minutes, for the timelapse of the previous print to be created.

#### Branch & Commit or Version of OctoPrint
Version: 1.3.1 (master branch)

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Geeetech GS2 Pro - Marlin

markwal, 2017-03-29:
You can do this already if you wrap avconv in a script that uses nice.  But I think CPU isn't the only contention problem.

JohnOCFII, 2017-03-29:
@fast930 - what type of Pi do you have?  I wonder if this warning is still pertinent with the available power of a Pi 3, for example.  Looking at top when a time lapse is running, it would seem there are plenty of resources available to start a print.

markwal, 2017-03-29:
While its true that the Pi3 has multiple cores and OctoPrint can only use one, ffmpeg can use many, but you can set how many threads in the UI, but actually I don't think CPU on a pi 3 is your main contention problem.

I think OctoPrint still does the still captures to a single folder ~/.octoprint/timelapse/tmp and you can't capture new frames to that place while you're making a timelapse, at least not in a way that will work well because the ffmpeg command line specifies: make a timelapse moving out of all the stills in this folder.

fast930, 2017-03-29:
@JohnOCFII - I'm running it on a Pi B because that's what I had sitting around and it met the system requirements.  So, it's a single core CPU at 700mhz, but despite that it seems to work just fine.  

When I have been bold enough to start a print while a timelapse is still rendering I haven't noticed any impact on the resulting print.  But I haven't measured print times with/without a timelapse rendering or printed anything complex that might be more likely to be impacted by Octoprint not keeping up with the printer.  You may be correct that there are sufficient resources (even on my little Pi B) to handle printing and rendering at the same time, in which case the warning wouldn't be pertinent.  And I'm doing my slicing elsewhere - I could imagine that the combination of printing, rendering and slicing combined would give that Pi B a workout.

@markwal - I realize I could wrap avconv in a script that uses nice.  In fact, I've used renice from the command line to drop the priority of avconv a couple of times when I was itching to start another print but a timelapse was still rendering.  I guess what I was envisioning was a place in the UI that would allow the user to give timelapse rendering a lower priority.  Or conversely, give octoprint a higher priority which would give it priority over both avconv and possibly a slicer, etc.?

You suggest that maybe CPU isn't the only contention.  Good point - I could imagine disk I/O being a bottleneck in some situations with rendering, especially for someone running on a slow SD card in their Pi.  Are there other use cases that you have in mind that aren't CPU bound?

At the end of the day, it's about making the printing function as close to real-time as possible to keep the printers firmware buffer full and avoiding any pauses in printing that affect print quality.  Seems to me that a relatively simple way to improve the odds would be to make use of the OS provided CPU priority.  This is something that could be done on all supported OS's but may only impact configurations at the low end of the system requirements.  And, maybe it's moot as @JohnOCFII suggests, even with a lowly Pi B, in which case the message could be ignored or removed.

fast930, 2017-03-29:
@markwal - it looks like the files in ~/.octoprint/timelapse/tmp are unique, having a print object name, date/timestamp and sequence number.  What about this would prevent a capture from proceeding while a rendering is in progress?  Wouldn't ffmpg / avconv act only on files with a single, unique print object name and time/datestamp for a completed print?

foosel, 2017-03-29:
The problem with changing process priority is that OctoPrint is platform agnostic. Windows doesn't have nice. So a cross platform solution needs to be found. And "if windows do foo else bar" branches in the code tend to decrease maintainability and testability. I already have a bunch of those in there, and every time I have to add a new one I cringe. So keep that in mind when asking for features as this one - some stuff is better solved outside of OctoPrint because it is a heck of a lot easier when you don't have to keep platform agnosticism in the back of your head all the time. Not saying that I won't take a look if this is something we could still add, just saying that if you want a fast solution better go for the wrapper script.

@markwal btw is right in that the timelapse temp folder used to behave like that, but that was changed around last summer or something to prevent people from nuking their ongoing render jobs just be starting the next print job.

fast930, 2017-03-29:
@foosel - Thanks for looking at this.  Your point about the added complexity of implementing this in a cross platform way is well taken.  Although it's possible in all supported OS's I believe, Windows is going to be the biggest difference and I understand why you cringe at adding another "if windows...".

For the time being, I'm probably going to wrap avconv in a script that sets the nice value to a lower priority.  Or perhaps I'll look at setting python's execution of octoprint to a *higher* priority so that rendering + anything else I do in the future (slicing, etc.) doesn't impact printing.

BTW - good discussion about cross-platform process priority in Python here: http://stackoverflow.com/questions/1023038/change-process-priority-in-python-cross-platform

Thanks again for considering this and for all your fantastic work on Octoprint!

ntoff, 2017-03-30:
What about an option to just not render the timelapse? There are buttons on the unrendered timelapses so you can render them, why not a tickbox in the timelapse settings for "capture only, render later" and users can just queue up the renders using their preferred method (plugin, batch file, bash script, etc) or manually hit the render button? Not really an issue I've ever faced as I have to let my bed cool to get the prints off, then heat back up again and usually by then the rendering is done but I can see people who print on unheated beds might have a faster turnover.

foosel, 2017-03-30:
@ntoff that would be what #1313 would be about :)

ntoff, 2017-03-30:
typical me, always behind the times :(

foosel, 2017-03-30:
To be fair, I wouldn't have found that ticket that quickly if I hadn't just clustered all open feature requests using a huge spreadsheet I can now very quickly look up stuff in ;)

jgarland79, 2017-10-01:
This could also be applied to the local slicer so that it does not need to be disabled while printing. Just set the nice level on the slicer so that is has a lower priority.

fiveangle, 2017-10-02:
While it can't be easily solved for all platforms, it probably makes a lot of sense to submit a ticket to OctoPi to tweak the defaults to renice launched ffmpeg instances to +19, haproxy to +15.  For octoprint process, renice on single-core to -5 and on anything > 1 core -19.

Sophist-UK, 2020-12-30:
> @fast930 - what type of Pi do you have? I wonder if this warning is still pertinent with the available power of a Pi 3, for example. Looking at top when a time lapse is running, it would seem there are plenty of resources available to start a print.

The lower the power, the more important this is - but preventing stutters is really pretty important.

fiveangle, 2020-12-31:
btw- Not to claim there is a solution for all platforms, but Windows does have a process priority feature equivalent in the form of the built-in "start" command.

I don't know if the mechanism is drop-in for how OP launches child encoding processes, although I would imagine Gina has full control over that 🤷

```
C:\Users\speedster>start /?
Starts a separate window to run a specified program or command.

START ["title"] [/D path] [/I] [/MIN] [/MAX] [/SEPARATE | /SHARED]
      [/LOW | /NORMAL | /HIGH | /REALTIME | /ABOVENORMAL | /BELOWNORMAL]
      [/NODE <NUMA node>] [/AFFINITY <hex affinity mask>] [/WAIT] [/B]
      [command/program] [parameters]

    "title"     Title to display in window title bar.
    path        Starting directory.
    B           Start application without creating a new window. The
                application has ^C handling ignored. Unless the application
                enables ^C processing, ^Break is the only way to interrupt
                the application.
    I           The new environment will be the original environment passed
                to the cmd.exe and not the current environment.
    MIN         Start window minimized.
    MAX         Start window maximized.
    SEPARATE    Start 16-bit Windows program in separate memory space.
    SHARED      Start 16-bit Windows program in shared memory space.
    LOW         Start application in the IDLE priority class.
    NORMAL      Start application in the NORMAL priority class.
    HIGH        Start application in the HIGH priority class.
    REALTIME    Start application in the REALTIME priority class.
    ABOVENORMAL Start application in the ABOVENORMAL priority class.
    BELOWNORMAL Start application in the BELOWNORMAL priority class.
    NODE        Specifies the preferred Non-Uniform Memory Architecture (NUMA)
                node as a decimal integer.
    AFFINITY    Specifies the processor affinity mask as a hexadecimal number.
                The process is restricted to running on these processors.

                The affinity mask is interpreted differently when /AFFINITY and
                /NODE are combined.  Specify the affinity mask as if the NUMA
                node's processor mask is right shifted to begin at bit zero.
                The process is restricted to running on those processors in
                common between the specified affinity mask and the NUMA node.
                If no processors are in common, the process is restricted to
                running on the specified NUMA node.
    WAIT        Start application and wait for it to terminate.
    command/program
                If it is an internal cmd command or a batch file then
                the command processor is run with the /K switch to cmd.exe.
                This means that the window will remain after the command
                has been run.

                If it is not an internal cmd command or batch file then
                it is a program and will run as either a windowed application
                or a console application.

    parameters  These are the parameters passed to the command/program.

NOTE: The SEPARATE and SHARED options are not supported on 64-bit platforms.

Specifying /NODE allows processes to be created in a way that leverages memory
locality on NUMA systems.  For example, two processes that communicate with
each other heavily through shared memory can be created to share the same
preferred NUMA node in order to minimize memory latencies.  They allocate
memory from the same NUMA node when possible, and they are free to run on
processors outside the specified node.

    start /NODE 1 application1.exe
    start /NODE 1 application2.exe

These two processes can be further constrained to run on specific processors
within the same NUMA node.  In the following example, application1 runs on the
low-order two processors of the node, while application2 runs on the next two
processors of the node.  This example assumes the specified node has at least
four logical processors.  Note that the node number can be changed to any valid
node number for that computer without having to change the affinity mask.

    start /NODE 1 /AFFINITY 0x3 application1.exe
    start /NODE 1 /AFFINITY 0xc application2.exe

If Command Extensions are enabled, external command invocation
through the command line or the START command changes as follows:

non-executable files may be invoked through their file association just
    by typing the name of the file as a command.  (e.g.  WORD.DOC would
    launch the application associated with the .DOC file extension).
    See the ASSOC and FTYPE commands for how to create these
    associations from within a command script.

When executing an application that is a 32-bit GUI application, CMD.EXE
    does not wait for the application to terminate before returning to
    the command prompt.  This new behavior does NOT occur if executing
    within a command script.

When executing a command line whose first token is the string "CMD "
    without an extension or path qualifier, then "CMD" is replaced with
    the value of the COMSPEC variable.  This prevents picking up CMD.EXE
    from the current directory.

When executing a command line whose first token does NOT contain an
    extension, then CMD.EXE uses the value of the PATHEXT
    environment variable to determine which extensions to look for
    and in what order.  The default value for the PATHEXT variable
    is:

        .COM;.EXE;.BAT;.CMD

    Notice the syntax is the same as the PATH variable, with
    semicolons separating the different elements.

When searching for an executable, if there is no match on any extension,
then looks to see if the name matches a directory name.  If it does, the
START command launches the Explorer on that path.  If done from the
command line, it is the equivalent to doing a CD /D to that path.
```

Sophist-UK, 2020-12-31:
I use a tool called System Explorer on windows which provides a Task Manager equivalent/replacement, but also allows you to set persistent process priorities based on process name. So you can set all instances of octoprint.exe to Above Normal for example.

I have (literally) just configured the service definition on my new Raspberry Pi 400 octoprint server (which I can heartily recommend!!!) as follows:
```
[Service]
Environment="LC_ALL=C.UTF-8"
Environment="LANG=C.UTF-8"
Type=exec
User=octoprint
ExecStart=/opt/OctoPrint/bin/octoprint
Nice=-2
IOSchedulingPriority=2
```