OctoPrint/OctoPrint #1845: Error while uploading to the OctoPrint server: 500 (slic3r  Prusa Edition)

MartinMajewski, 2017-03-31:
As I have no idea if this issue is OctoPrint or Prusa related, I will post it in both repositories.

#### What were you doing?

Hitting the upload button in slic3r 1.34.0 Prusa Edition results in the error shown in the screenshow attached.

#### What did you expect to happen and what happened instead?

No error message.

#### Branch & Commit or Version of OctoPrint

Slic3r: 1.34.0
OctoPrint: 1.3.2 (rc/maintenance branch)

#### Operating System running OctoPrint

Windows 10 x64

#### Printer model & used firmware incl. version

Original Prusa i3 MK2

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 57.0.2987.110

#### Link to octoprint.log

No log created by OctoPrint for this issue

#### Screenshot(s) or video(s) showing the problem:

[image]

foosel, 2017-03-31:
> No log created by OctoPrint for this issue

Then it didn't even reach the OctoPrint server and the problem arises somewhere in between.

Try uploading the same file through OctoPrint's interface. If that works, the issue lies with Slic3r.

MartinMajewski, 2017-03-31:
Jap... seems to be an slic3r issue with the most current 1.34.0 version!

Thank you for the response!

foosel, 2017-03-31:
Closing this then as it's not an OctoPrint issue. Cheers!

closed 2017-03-31 by foosel