OctoPrint/OctoPrint #1846: HaProxy tremendously slowing down Webcam Stream Performance

RodneyMcKay715, 2017-03-31:
#### What were you doing?

I installed OctoPrint 1.3.2 (master branch) and installed HaProxy alongside to make my URL a bit nicer and control everything with 1 Portforwarding on my Router.
My OctoPrint has the Adress of the Pi (format: 192.168.1.1)
My Camera has the Address    /webcam/?action=stream

#### What did you expect to happen?

That i can view the Webcam at 720p and 25 fps normally without any lags or stutter

#### What happened instead?

The Webcam is unresponsive and freezing and laggy with a delay of ca. 10 seconds - 20 seconds when used over haproxy. (My Upload is good enough to handle it).
Accessing the Camera over the Direct Ip of the Stream makes it very fluent and without any stutter.
The CPU usage of my Raspberry Pi also never exceeds 5% at all times.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch) 

#### Printer model & used firmware incl. version

Anet 3D A6 (firmware version not available)

#### Browser and Version of Browser, Operating System running Browser

Firefox 52.0.1 without any addons, Windows 7 64x

#### Screenshot(s) showing the problem:

I created 2 youtube videos.
One with the Usage of HaProxy:
https://www.youtube.com/watch?v=R9JRTMopaWg

The Other without HaProxy over direct Link:
https://www.youtube.com/watch?v=bB5B771TQFs

I have read the FAQ.

foosel, 2017-04-03:
Sorry, but there is really nothing I can do about this since I do not have any control over how haproxy performs, or its implementation details, or anything else related to it for that matter. I'm merely using it myself and since recommend to do so since it works for me. I am seeing a bit of lag in my own setups from time to time, but I personally can live with that.

You could look if another reverse proxy (nginx, apache with mod_proxy) performs better for you, but that's about the only advice I can offer here. There are [configuration snippets](https://github.com/foosel/OctoPrint/wiki/Reverse-proxy-configuration-examples) on the wiki.

> My Upload is good enough to handle it

Wait, so have you tested in your local network too, or only through your forwarded port? If the latter I'd suggest to take the port forward out of the equation and test in your LAN if the same symptoms appear there too.

> It also needs to be noted that following this step in the instructions: [server bound to 127.0.0.1 only] Causes my Raspberry Pi to reboot on Webcam Access.

This line does nothing else but tell OctoPrint to only bind to the loopback address (127.0.0.1) instead of all available network addresses and has nothing to do at all with whether there is a webcam server running in an entirely different process on the same machine or not. If your Pi reboots when you add this line, something is wrong with either your hardware or your system integrity. Reboots out of the blue when utilizing the system reek of the PSU not being sufficient for the application (the Pi is notoriously bad when it comes to insufficient power supplies and shows hilarious brown out issues in such cases that are hard to debug). Make sure your PSU can deliver *continuous* 2A (2.5A for a Pi3!) at 5V (not less than that). Phone/tablet chargers are usually quite prone to performing in a suboptimal way (even if they claim to deliver the mentioned amps/voltage).

closed 2017-04-03 by foosel

RodneyMcKay715, 2017-04-03:
I got it working (mostly).
If i start octoprint and use a wired connection there is an issue however. Sometimes everything works fine, little to no delay in the camera, but sometimes it just starts to lag horribly. At first i thought it was the Pi itself, but i dont know. If it starts to lag, i get the Popup, that a new server version has been detected and the pi starts over and over again. The other scenario is, where this doesnt happen that the camera is delayed 15-20 seconds, like in the Video shown in my Github issue report. The last scenario is that Octoprint just says "The Server has been shutdown" if i click any Menu Entry.
The funny thing is however, that if i access the stream directly over <external ip>/webcam/?action=stream its fluid and has no issues. As soon as i use it in combination with the Web interface the Interface begins to behave delayed and the stream starts to lag.Also it needs about 30 seconds to fully load up (tested without proxy as well.) and even that is not guaranteed. Sometimes it just loads the upper bar and thats it, other times its loaded instantly.If it doesnt load the Pi is still fully accessible over SSH. Is there any way to fix this issues? My router is a FritzBox 3270 , my PI is an Raspberry PI 3 Model B not overclocked, using only the command line. What  is weird however is that it barely uses any CPU capacities. It uses 0 - 3% cpu if running the camera stream (original raspi cam) and octoprint.
This issues happen with Raspbian Lite, Raspbian+ Pixel, OctoPi and an manual OctoPrint installation.
Looking forward to hear from you,

nophead, 2017-04-04:
I think the issue is simply that haproxy adds buffering, so if the camera produces data faster than it gets away on the network the buffer fills up and you get a big lag. When mjpg-streamer is writing to the network directly it presumably drops frames if the network doesn't keep up as it makes no sense to buffer live video.

nophead, 2017-04-04:
Is it possible to not route the camera through haproxy but still make it appear in the control tab?

jneilliii, 2017-04-05:
Have you tried the SnapStream plugin yet to see if the issue would still exist?  http://plugins.octoprint.org/plugins/snapstream/

foosel, 2017-04-05:
> Is it possible to not route the camera through haproxy but still make it appear in the control tab?

Of course, just use the direct URL to the webcam server as the stream URL (`http://<pi's IP>:8080/?action=stream` on OctoPi). 

Since there does not exist any way to specify "port relative" URLs though (something like "resolve this relative to my protocol and host name but with this port and path"), you will need to hardcode a hostname or IP for this to work, which makes it impossible to use that for multi homed Pi's (e.g. ip vs hostname, or in a port forward scenario). Which is the reason why the use of haproxy (or generally a reverse proxy of any kind) is recommended here, it simply makes stuff like this way way less of a headache or even possible at all.

nophead, 2017-04-05:
So when I show the direct stream side by side with the haproxy version I can see that haproxy is causing the lag. It varies between about 2s and 20s seemingly randomly. I don't suppose haproxy has an option not to buffer?

foosel, 2017-04-05:
Whatever it does with regards to buffering/rechunking can apparently be [switched off via the `http-no-delay` option](http://marc.info/?l=haproxy&m=131102278201729&w=2) but in my experiments that didn't help at all. So either I misunderstood this option, or buffering/rechunking of the response is not the problem here. Your mileage might vary though, so you might want to give it a shot.

I had to ramp up my webcam to 30fps to be able to reproduce the issue, and then it reproduced equally "well" on haproxy and nginx, so it's not a haproxy specific problem. MJPG is what it is - a constant stream of JPEGs. Without any kind of protocol inspection (that would add additional overhead) there's no way for a reverse proxy to drop frames if the client connection can't keep up. Maybe there are solutions out there better than haproxy and nginx for this job - I'm happy for suggestions. I'm also here and there looking into getting away from MJPG towards something more suitable for streaming (the browser support is thankfully there by now), but that's currently more a case of "shower thoughts" considering that I already have my hands full with everything else. 

I do not get reproduction at all when running my webcam at 720p and 10fps (over WiFi even), and in my book that is completely sufficient for remote monitoring. But if that is not acceptable for you, I suggest to get on with the experimenting ;)

nophead, 2017-04-05:
I don't think the no delay option is relevant here. It seems to mean it stops it waiting for a certain amount of data before sending it out. I don't think that is what the problem is here. It looks like haproxy accepts data from mjpg-streamer faster than it can send it out again and builds up a long queue of data.

I am only trying to stream 640x480 at 4 fps. I would love 720p and 10fps but I have never been able to achieve that with an RPI camera without using too much CPU time.

I don't know how you get more throughput on WiFi than I do. I even bought the Edimax Wireless N150 Nano, which I think is what you said you used. WiFi dongles with tiny aerials never perform well in my house, only devices with proper full size aerials.

I am just hard wiring my last two machines at the moment. I have a 16 port switch at one side of my lab and an 8 port switch at the other, an 8 port switch in my garage workshop and a five port switch in my lounge. It's crazy but I only use Wifi for mobile devices like phones and laptops.

RodneyMcKay715, 2017-04-05:
I tried the OctoPi installation but my router is going completely crazy then. If i connect my pi to the power my router gets 4 devices named octopi and in the octopi network txt there are 4 netowrk adaptors defined , why is this?

koenkooi, 2018-01-28:
This needs more testing, but I get the impression the lag is a lot better with the following change in nginx.conf. But it could very well be the placebo effect:

```
diff --git a/recipes-webui/octoprint/octoprint-nginx/nginx.conf b/recipes-webui/octoprint/octoprint-nginx/nginx.conf
index 3e33b79..4f9e3a4 100644
--- a/recipes-webui/octoprint/octoprint-nginx/nginx.conf
+++ b/recipes-webui/octoprint/octoprint-nginx/nginx.conf
@@ -43,6 +43,7 @@ http {
 
         location /webcam/ {
             proxy_pass http://mjpg-streamer/;
+            proxy_request_buffering off;
         }
 
         # redirect server error pages to the static page /50x.html
```

shivan, 2018-03-25:
Any news here? I'm experiencing the same video lag of about 10 seconds in the OctoPrint UI. 

Using the direct stream:
http://octopi:8080/?action=stream
it is fast as expected (15fps) - 1280x720

But on the UI of octoprint itself I've got a lag of about 10 seconds.

Using PI3 B+
Raspi camera

Browser: Chrome (latest)

Even when entering http://octopi:8080/?action=stream as camera address, it is the same slow.
Using default settings of 640x480 10fps it is slow in OctoPrint UI, too.

OctoPi version: 2018-03-15_2018-03-13-octopi-stretch-lite-0.15.0 (because 0.14 does not work on PI3 B+)

Blu-Eagle, 2018-10-04:
you'll get rid off delay if in the webcam stream field you will use:

http://octopi3.local:8080/?action=stream

of course octopi3.local is the name of the octopi on the network just use yours ;)