OctoPrint/OctoPrint #1848: [Request] timelaps capture point idea

stijnpeeters, 2017-04-01:
the timelaps is now captured at z level or time. but what if it was synct to the center of highest point of the object and every time the nozzle passes this center (and z is 1 up)... then the timelapsmovie would have the object centered at each frame in the eventual movie. This will make for a much more beautiful movie because the object appears to stand still and grow.

ntoff, 2017-04-01:
Octoprint doesn't know where your printer is, it could read the gcode sent to the printer (provided you're not printing from SD) but your printer doesn't necessarily execute the  move as soon as it gets the instruction. That's why if you hit the "Cancel" button, the printer may actually continue to print until its own buffer is empty. 

The way most people do this (I assume you have a Prusa style printer with a moving bed) is to attach the camera to the print bed.

stijnpeeters, 2017-04-01:
i am aware that it could be connected to the bed. but that add weight..thus
slows it down. I think it would be possible to calculate it based upon the
gcode. basicly its the gcode vieuwer averaged out seen the last layer.

foosel, 2017-04-03:
> basicly its the gcode vieuwer averaged out seen the last layer.

No, it isn't. There is a planner buffer in your firmware. It's **impossible** to know where your printer's head is at a given point in time, at least accurately enough to trigger a timelapse snapshot only then to ensure "wobble free" timelapse recordings.

> I think it would be possible to calculate it based upon the gcode.

No, it wouldn't, if by "it" you mean "the exact point in time when a snapshot would need to be triggered to capture the print head ad exactly the right position". But I'd be happy to be proven wrong, the source is there for you to experiment on.

What you are asking for here is simply not possible with how firmware and printers currently work. You could "get close", but it still would run into the exact same issues as the current implementation, just eat a ton of more resource in the process, only work for local files AND necessitate a full model analysis beforehand, so no quick upload and print cycle possible at all. It simply doesn't make "economical" sense.

Closing this as "won't implement".

closed 2017-04-03 by foosel