OctoPrint/OctoPrint #1850: Plug-in manager disappeared after update.

KevMags, 2017-04-01:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete any lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to repor tis still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the contribution
guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

After upgrading, the plug-in manager is no longer in the settings window.  (3 different machines, all the same thing)  

#### What did you expect to happen and what happened instead?
Expected access to Plug-in manager to add some of the cool new plug-ins.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.3.dev40+gd72c7c1 (maintenance branch)

#### Operating System running OctoPrint

macOS sierra 10.12.5/Jessie Lite on Pi3.

#### Printer model & used firmware incl. version

Voron, Marlin RCBugFix (March 20)

#### Browser and Version of Browser, Operating System running Browser

Safari 10.1.1, macOS Sierra 10.12.5

#### Link to octoprint.log

[octoprint.log.2017-03-31.html.txt](https://github.com/foosel/OctoPrint/files/887790/octoprint.log.2017-03-31.html.txt)
[octoprint.log-2.html.txt](https://github.com/foosel/OctoPrint/files/887791/octoprint.log-2.html.txt)

#### Link to contents of terminal tab or serial.log
[TerminalTab.txt](https://github.com/foosel/OctoPrint/files/887795/TerminalTab.txt)

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting any kind of communication issues between OctoPrint and your printer.
Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser
[JavaConsole.txt](https://github.com/foosel/OctoPrint/files/887797/JavaConsole.txt)

#### Screenshot(s) or video(s) showing the problem:
[Screen Shot 2017-04-01 at 12.44.16 PM.png.txt](https://github.com/foosel/OctoPrint/files/887798/Screen.Shot.2017-04-01.at.12.44.16.PM.png.txt)
I'm not sure if this will work, I added ".txt" to the end.

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.
Yep

KevMags, 2017-04-01:
Just a note, probably does not apply.  The beta OctoPi image downloaded yesterday (OctoPrint 1.3.2 (master branch)) is not affected by this.

ntoff, 2017-04-01:
Seems fine here running on a windows 7 PC (p.s. @KevMags to post a screenshot, just copy it to the clipboard and paste it) although I did it manually via `git pull / checkout maintenance / python setup.py develop` so whether that changes anything I don't know. Maybe try a manual upgrade? Are the machines you upgraded an old octopi install or were they manually set up?

plugin manager:
[image]

and version number:
[image]

KevMags, 2017-04-01:
Is it possible to manually install just the plug-in manager?

KevMags, 2017-04-01:
Clean install from Octoprint image (@octoprint.org), updated using the following:

cd ~/OctoPrint
source ~/oprint/bin/activate
git fetch
git checkout maintenance
python setup.py clean
python setup.py install
sudo service octoprint restart

No plug-in manager in settings menu.
Internet Explorer w/Win8 or Mac Safari.

[ScreenShot.png.txt](https://github.com/foosel/OctoPrint/files/888040/ScreenShot.png.txt)
[Screen ShotMacSafari.png.txt](https://github.com/foosel/OctoPrint/files/888042/Screen.ShotMacSafari.png.txt)

(remove .txt and view as PNG file)

ntoff, 2017-04-01:
You don't need to add .txt to your screenshots, you can just upload them directly by pasting them into the reply box. Makes it a lot easier for us to look at them not having to download and rename them.

Instead if git fetch, try git pull

KevMags, 2017-04-01:
Sorry, I'm not exactly a Rocket Surgeon with GitHub.   When trying "pull" it says I'm  up to date.  Worked before the update, now not working.  I can copy an image that I have that has the plug-ins installed and edit from there to get things going for the other printers or punt.  I will wait for the 1.3.3 release and try again.  They work fine otherwise.

ntoff, 2017-04-02:
I'm no rocket scientist either but according to https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image

the correct command is git pull.

https://github.com/freeCodeCamp/freeCodeCamp/wiki/Git-Pull-Vs-Git-Fetch

>git fetch is the command that tells your local git to retrieve the latest meta-data info from the original (yet doesn't do any file transfering. It's more like just checking to see if there are any changes available)

Confirmed that when using `git fetch` the plugin manager doesn't load, however `git pull` results in a proper install and the plugin manager is present.

use `git pull` as per the official instructions, and it works.

KevMags, 2017-04-02:
OK, a fresh start on a day that my hair is not on fire.  Retried using pull and things are going good.  Thanks for helping.

foosel, 2017-04-03:
```
2017-04-01 08:50:59,693 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.3.dev40_gd72c7c1-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-04-01 08:51:00,993 - octoprint.plugin.core - ERROR - Error loading plugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.3.dev40_gd72c7c1-py2.7.egg/octoprint/plugin/core.py", line 685, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.3.dev40_gd72c7c1-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 29, in <module>
    import dateutil.parser
ImportError: No module named dateutil.parser
2017-04-01 08:51:01,357 - octoprint.plugin.core - INFO - Found 14 plugin(s) providing 14 mixin implementations, 8 hook handlers
```

Forgot to add a dependency, my bad, fix incoming.

---

As a side note, are you intentionally running the `maintenance` **branch** or did you actually want to subscribe to the "Maintenance RC" **release channel**? Those are not the same, the `maintenance` branch is basically the development version of the next release candidate and will report an update available basically every time I push a new commit there (worst case: multiple times per day). The "Maintenance RC" release branch is what you want if you want a bit less update intense experience (also with less risk of intermediary breakage - shouldn't happen usually, but still can, it's a development version).

foosel, 2017-04-03:
Fixed again, thanks for reporting.

commit 12199391 (2017-04-03): Add missing dependency

Closes #1850
closed 2017-04-03 by foosel