OctoPrint/OctoPrint #1853: Print Stop Mid Print, random times on different models

BryanArnold, 2017-04-05:
#### What were you doing?

I printed with Octoprint successfully for months now, and the last 2 models stopped mid-print, at random times, multiple times. I downloaded the gcode file and checked it out, should print to the end. I am hooked up USB, powered by a 5.2V power supply.

#### What did you expect to happen and what happened instead?

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Prusa i3 MK2

#### Browser and Version of Browser, Operating System running Browser

Not browser dependant, using Chrome though.

#### Link to octoprint.log

https://pastebin.com/09M4qy12

#### Link to contents of terminal tab or serial.log

serial.log was not enabled, and I could not get into the interface after the print had stopped, it was locked up and not responding.

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) or video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2017-04-05:
> and I could not get into the interface after the print had stopped, it was locked up and not responding.

That sounds like the whole Pi crashed then, which would not be an OctoPrint problem. It would be interesting if you could still ping your Pi in that situation (`ping octopi.local`, `ping <your pi's ip>`). If that didn't work anymore, it's definitely the whole Pi that was gone.

Your `octoprint.log` shows quite a number of uncontrolled restarts of the server like this:

```
2017-04-03 17:56:23,335 - octoprint.server.util.sockjs - INFO - Client connection closed: 71.11.132.25
2017-04-03 18:17:08,224 - octoprint.server - INFO - ******************************************************************************
2017-04-03 18:17:08,226 - octoprint.server - INFO - Starting OctoPrint 1.3.2 (master branch)
2017-04-03 18:17:08,226 - octoprint.server - INFO - ******************************************************************************
```

This is what it should look like when you shut down/restart the Pi properly:

```
2017-04-05 06:38:15,966 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.3
2017-04-05 06:38:16,215 - octoprint.server - INFO - Shutting down...
2017-04-05 06:38:16,306 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2017-04-05 06:38:16,314 - octoprint.events - INFO - Event loop shut down
2017-04-05 06:38:16,355 - octoprint.server - INFO - Goodbye!
2017-04-05 06:39:01,198 - octoprint.server - INFO - ******************************************************************************
2017-04-05 06:39:01,203 - octoprint.server - INFO - Starting OctoPrint 1.3.2 (master branch)
2017-04-05 06:39:01,205 - octoprint.server - INFO - ******************************************************************************
```

Note the "Shutting down..." etc before the "Starting OctoPrint" banner. That is missing in a lot of places in your log, so that either indicates that you are simply switching off your Pi instead of shutting it down properly (never do that, always shut it down properly, you risk serious data loss) or that either OctoPrint or your Pi spontaneously rebooted/crashed and had to be restarted. To differentiate between the two latter options, please take a look into `/var/log/syslog` and checking if there are any kernel messages in there that might help narrow this down.

Based on your description of the symptoms and the available logs however it definitely sounds like a stability issue with the system *under* OctoPrint, not anything OctoPrint could do anything about.

BryanArnold, 2017-04-05:
Wow, thanks for taking the time to write a detailed response!

That totally makes sense, it does seem like the whole system is freezing up. I never just turn off the power, I always shut down first. That is, unless it's frozen. That's why there so many of those in the log, that's the only way to get back into Octoprint.

Looked in the syslog, did not see anything that stood out as wrong.

I am powering the pi off of a converter, powered by the printers power supply. Even though that's set to 5.2V, I wonder if that power is not constant, and causing the freezes.

One question, would the USB cable going to the printer coming loose, or intermittent connection cause a pi to freeze up?

foosel, 2017-04-05:
It would certainly cause issues with your ongoing print, I can tell you that. Any kind of USB connect/disconnect can also always induce a voltage spike into the system, so who knows. I'd rather expect sudden reboots in such cases, not freezes.

If you have fluctuating voltage or your power connection can't provide the Pi with enough current (my personal rule of thumb: 2A for Pi2, 2.5A for Pi3), that certainly can cause all kinds of issues. The Pi is notoriously sensible to suboptimal power supply situations (symptoms include communication errors with printers, disconnections, weird network hiccups, outright reboots, and more random shenanigans).

As a side note, I'm marking this as closed because right now it really doesn't sound like an OctoPrint bug. Happy to reopen if we find reason to believe that this is wrong after all. Comments are still possible even on closed issues.

closed 2017-04-05 by foosel

BryanArnold, 2017-04-05:
Thanks.

The power supply is my current suspect. I have a standalone 2.5V power supply I will try to troubleshoot with.

foosel, 2017-04-05:
> I have a standalone 2.5V power supply

I hope you mean **5**V but 2.5**A** :)

BryanArnold, 2017-04-05:
LOL, oops. Yes 5V 2.5A

Erutan409, 2017-04-05:
I'm currently working on a stable power supply method.  @foosel, we've talked about this in a couple different issues, before.  I've [discovered that the reason my prints were stopping were due to irregular voltage on my circuit the raspberry pi/printer is plugged into](http://community.robo3d.com/index.php?threads/trying-to-figure-out-connectivity-issues.17987/#post-104949).

I can't say with 100% certainty that this is @BryanArnold's issue, but I'd guess it is.

I put it together after my miniature clone was testing the light switch in my office (where my setup is located) and witnessed my print fail.  So, after some testing and reproducing (logs + voltmeter + wasted filament), I'm confident that those wall wart's that are marketed for the raspberry pi don't do a great job in regulating the voltage in these cases.

So, I'm currently working on getting a switching power supply in the mix to solve the aforementioned voltage irregularity.  But, I'm hitting a blocker with understanding how/why the Arduino in my printer isn't particularly happy with the switching power supply.

Once I have it all figured out, I was going to share my findings and hopefully resolve this reoccurring reported issue for good; which, is not the fault of the OctoPrint software.

BryanArnold, 2017-04-05:
Thanks for the reply. The more I think about it, I agree with you about the power supply. I am using one of these powered by the Prusa i3 MK2 power supply.

https://www.amazon.com/gp/product/B00JUFJ1GA/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

It's supposed to be rated for up to 3V, but who knows how stable.

foosel, 2017-04-05:
Ampere, not Volt ;)

@Erutan409 I think I remember. It currently feels like the power issues are getting worse, I get more and more reports along those lines. Might be related to the Pi3 becoming the more common machine, stuff you could still get away with power wise with a Pi1 and Pi2 no longer flies with that one. Even had to swap one of my own supplies for the test cluster after I noticed power fluctuations. Very annoying issue ;)

nophead, 2017-04-05:
The link above only says 2A, not 3A.

I have used one of these: http://www.ebay.co.uk/itm/172432528752 to power
an RPI3 from 12V. It is 3A and produces very little ripple. A lot of cheap
power converters produce horrendous ripple. I don't know why the advert
says DEOK as they are actually marked DROK.

And I always feed the power into RPIs via the GPIO connector because micro
USB cables have too much resistance.

The red LED is useful on the RPI as it is connected to a low voltage
detector so it blinks when supply falls below 3.6V IIRC. The CPU should be
happy down to that voltage but any USB devices powered by the RPI will
flake out sooner.

BryanArnold, 2017-04-14:
Follow up after some testing...

I narrowed down the system lock up to a faulty USB cable. When I swapped the power supply to a quality Pi external supply, I still had the problem. When I swapped USB cables, the problem went away. 

I have done 4 long prints without a problem. 

I found it out by wiggling the USB cable, and watching the Octoprint interface. When the cable moved, the printer disconnected.