OctoPrint/OctoPrint #1857: [Request] disable cam/alternate stream

BETLOG, 2017-04-08:
[Request]
Option to disable camera stream.
Option to enter alternative cam stream source.

Because they would be cool.
Also because I suspect octo is being greedy with the camera control and causing things like:
```
pi@octopi:~ $ raspistill -o picam.jpg
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
```

ntoff, 2017-04-08:
Octoprint only embeds the stream, mjpg-streamer is what handles the actual camera

https://github.com/jacksonliam/mjpg-streamer

BETLOG, 2017-04-09:
Thanks. Command line solutions are good, but Octoprint would benefit from a simple GUI option. Checkbox or settings menu option.
I was reading that cvlc might offer a less delayed image/actual video... but it was when I tried to raspistill and video and got a cryptic error that it seemed like the obvious suggestion.

foosel, 2017-04-10:
> Octoprint would benefit from a simple GUI option

A generic, available-for-everyone GUI option for that simply wouldn't make sense because OctoPrint doesn't control the webcam server and it doesn't even have to necessarily run on the same system that OctoPrint is installed on. Some people even use OctoPrint with a dedicated IP camera.

OctoPrint however does allow you to add arbitrary commands to its system menu (the same one through which you also shutdown/restart OctoPrint). You can do that either through [editing `config.yaml`](http://docs.octoprint.org/en/master/configuration/config_yaml.html#system) or - probably way easier - installing the [System Command Editor plugin](http://plugins.octoprint.org/plugins/systemcommandeditor/).

> I suspect octo is being greedy with the camera control

It's not OctoPrint but **mjpg_streamer** that is being "greedy" here, and simply because it has to for technical reasons - you can't access the camera through more than one process at the same time, and since the camera is being streamed, that means exclusive access while the server is running. If you need a still image from the camera on OctoPi while the webcam server is running, fetch it from the webcam server at `http://127.0.0.1:8080/?action=snapshot` (e.g. `wget -O picam.jpg "http://127.0.0.1:8080/?action=snapshot"`). 

I'm closing this now since as mentioned, it doesn't make sense to add a general "webcam on/off" option to OctoPrint, and adding one yourself is easily accomplished with things that are already built-in.

closed 2017-04-10 by foosel

BETLOG, 2017-04-10:
> System Command Editor plugin.
Outstanding!
I bet these three words answer a heap of new user's questions/suggestions/requests :)

postspectacular, 2017-04-18:
Another reason why it'd be great to have an UI option to "disable" the stream is bandwidth usage. On my Type A having the Octoprint UI open causes a constant 1MB+/sec stream, impacting others on my LAN, regardless if the cam view is shown or not. Would it not be possible to at least remove the `#webcam_image` element from the DOM if the cam view is not shown (and then re-add it again later)?

postspectacular, 2017-04-18:
The `#webcam_button` has already seems to have a `toggle_webcam_view` click handler assigned (which I can't find in the source, though) - should be a good place to do the above mentioned DOM edits...

foosel, 2017-04-18:
@postspectacular when you are not on the "Controls" tab, the image should get unloaded for this exact reason. That has been part of OctoPrint since a long time (even before 1.2.0? I'm unsure). 

That recently had to be changed though *for Safari only* because Safari would not stop streaming even on an unload, and consecutive resume of streaming would add more and more active streams. So if you are running Safari, that might explain your observation. Otherwise, it sounds like a bug.

postspectacular, 2017-04-18:
Thanks @foosel - but am using latest stable Chrome... TBH am not quite sure how recent the version bundled with the TAM Series 1 Pro is. It only shows a git SHA (ed683608beb9315fef672e3c7aa35a236153c39a) as version - could be ancient... will check with their support too!

foosel, 2017-04-18:
Huh, that SHA doesn't even exist in the repository, so I'm at a loss. Also not aware of any `#webcam_button` elements. Sounds like a somewhat modified version, maybe they removed/didn't add that particular behaviour.

postspectacular, 2017-04-18:
Thought so too by now - let's see what they say...