OctoPrint/OctoPrint #1859: Prints Randomly Fail Out

TheCatMan2000, 2017-04-10:
_What were you doing?_

I was trying to print something that took ~6 hours to print.

_What did you expect to happen and what happened instead?_

Expected: For this print to finish like all the other small prints.

What happened: Octoprint errors out with the state "Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1792" and the print is left unfinished.

_Branch & Commit or Version of OctoPrint_

OctoPrint 1.3.2 (master branch)

_Operating System running OctoPrint_

OctoPi 0.13

_Printer model & used firmware incl. version_

Monoprice Maker Select V2
I used whatever firmware came with it, I have no clue how to update it, or how to find out what version it is...

_Browser and Version of Browser, Operating System running Browser_

Windows 10, Google Chrome Version 57.0.2987.133

_Link to octoprint.log_

https://pastebin.com/Jrb45QiC

_Link to contents of terminal tab or serial.log_

https://pastebin.com/ndHdgfBu
This didn't happen at the same time, but it gave the same error.

_Link to contents of Javascript console in the browser_

N/A

_Screenshot(s) or video(s) showing the problem_

[image]

I have read the FAQ.

foosel, 2017-04-10:
Not an OctoPrint issue but something going wrong a layer below it. Most commonly caused by faulty power supplies on Raspberry Pis and/or broken or unsuited USB cables. See also #1603 and [the various other tickets](https://github.com/foosel/OctoPrint/issues?utf8=%E2%9C%93&q=is%3Aissue%20device%20reports%20readiness%20to%20read%20but%20returned%20no%20data) and also [this shiny new FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception).

closed 2017-04-10 by foosel

a0s, 2017-04-22:
Get same error on Marlin 1.1.0-RC8 and  OctoPrint 1.4.0.dev171
In dmesg
```
Apr 22 09:07:16 localhost kernel: [ 4421.260794] usb usb4-port1: disabled by hub (EMI?), re-enabling...
Apr 22 09:07:16 localhost kernel: [ 4421.267945] usb 4-1: USB disconnect, device number 2
Apr 22 09:07:16 localhost kernel: [ 4421.269258] cdc_acm 4-1:1.0: failed to set dtr/rts
```

a0s, 2017-04-24:
@foosel, you are right. I changed firmware of my board to the latest RCBugFix and error gone. Sorry)