OctoPrint/OctoPrint #1862: Terminal enableFancyFunctionality gets disabled when processing is too slow (like a tab is in the background), but never gets re-enabled

pyjamasam, 2017-04-11:
Just adding this issue here as we spoke about it a little while back on IRC, but I wanted to make sure it got logged so it didn't get forgotten (and I haven't noticed a commit to reflect it being fixed - so this is also to be used as a reminder :-) )

Performance is never evaluated again after fancyFunctionality is disabled for the terminal.  So once it gets disabled it never gets re-enabled till you reload the entire interface.

And at least under chrome when the Octoprint tab is in the background it often gets dinged performance wise so the processing time of log messages while printing some times drops below the acceptable limit.  But when the tab is back in the foreground performance increases again to well within acceptable limits, but sadly the terminal is still in dumb mode :-(

pyjamasam, 2017-04-11:
Updated original ticket so the bot is happier.

foosel, 2017-04-11:
Ah, perfect. While I now remember the conversation, thanks for adding a ticket, because of course I had forgotten about the issue again.

foosel, 2017-04-20:
Should be fixed now on `maintenance` (see above commit) and soon `devel`, will got in to 1.3.3.

Difficult to test though, I wasn't able to get my clients slow enough without using breakpoints in the javascript. Any additional "real world" tests against the `maintenance` branch are definitely welcome ;)

pyjamasam, 2017-04-20:
I'll give it a whirl.  I was able to see it fairly reliably with my configuration.

pyjamasam, 2017-04-24:
So far so good.  Seams to flip back and forth as needed.  haven't tried it during printing, but just having it spit out 105's is enough to see it.

A good way to test also it so throw a breakpoint into the log message processing stack somewhere.  Just sit on it for a second and then continue.  Then remove (or disable the breakpoint).

Temporarily causes log processing to take a while and then return to normal speed.

pyjamasam, 2017-04-26:
Ya after playing, this seems to have fixed the tab performance issues, but now auto scrolling is broken (as referenced in that commit I commented on) in that branch.

foosel, 2017-05-02:
@pyjamasam I was out of a commission for the past week. Just now took a look (using the same breakpoint approach that you described, which I actually also used during development testing), but I can't reproduce the "scrolling stops" behaviour right now over here. Did this only happen during actual printing or did you also repro this with the breakpoint approach?

pyjamasam, 2017-05-05:
No worries @foosel.

So the funny terminal behaviour some times manifests its self as the auto scrolling not working, and sometimes it just ends up being slightly "off" when scrolling.
Of course when i tried to actually replicate the issue I saw the strange off scrolling and not the autoscrolling being disabled.
https://youtu.be/7rJ3xeefALI

I have a personal plugin that just adds some CSS to make the terminal larger and the font smaller, but otherwise doesn't change anything else.  The issue is also visible (in my opinion actually looks worse) with the stock styles though.
https://youtu.be/JkRA2Tqi4vk

The other part of this is that if I add some debug logging as the first line in TerminalViewModel.scrollToEnd it doesn't ever get called.

Once I rename scrollToEnd to scrollToEnd2 (and update the calling methods to reference this new name) then I see the debug logging as well as better auto scrolling.
https://youtu.be/FwDCCoEjjlc

Hope that helps.

BillyBlaze, 2017-05-08:
Thanks for the info @pyjamasam 

Do you perhaps have any 3rd party plugins installed that could possible overwrite the scrollToEnd function?

pyjamasam, 2017-05-09:
My apologies for not catching that.  Yes it looks like I do have something that overwrites the terminal's scrollToEnd method.

Autoscroll (0.0.2)
https://github.com/MoonshineSG/OctoPrint-Autoscroll

https://github.com/MoonshineSG/OctoPrint-Autoscroll/blob/master/octoprint_autoscroll/static/js/autoscroll.js#L18

I can confirm disabling that plugin makes the issue go away.
Sorry about that.

Only thing I have noticed now (and it might be a side effect of using the background animation stuff to do the scrolling is that if I click on another chrome tab and leave octoprint in an inactive tab with the terminal visible for a little bit and then click back to octoprint's tab the auto scrolling hasn't happened and the terminal text is off the bottom.  As soon as another log message comes in the auto scrolling is done and the display looks correct.  Should never be an issue when priting due to the speed at which log messages come in.  But I notice it when just getting M105 requests and responses when idle.

I don't think its anything that needs to be fixed.  Just an observation.

I'll open a ticket on the autoscroll repo to deal with this as its whats broke.

I think this ticket can be closed as the original issue has been corrected. (well I guess close it when the release is released :-) )

BillyBlaze, 2017-05-10:
No problem, we thought this commit would have a low impact, but it seems it is actually has a very high impact. Also based on another problem @CapnBry found. We decided to revert this commit made by me and keep on using the old foreach binding.

foosel, 2017-05-10:
Yep, all reverted now. Good that this was identified as troublesome though before the first RC went out, less release overhead this way ;)

MoonshineSG, 2017-05-11:
I fixed the plugin (even if the OctoPrint change has now been reverted) and should not be causing issues if the function gets updated in the future ... 
sorry about causing the mess...

foosel, 2017-05-11:
@MoonshineSG I wouldn't call it a mess - it was only a development version that was affected and while it caused some head scratching, it did that to a limited number of people :) No harm done, and the change causing the issue caused other problems as well.

MoonshineSG, 2017-05-11:
ya, the beauty of having people willing to live on the edge... glad it was caught early.

foosel, 2017-05-31:
Oh, I should have closed this a while ago I think :)

closed 2017-05-31 by foosel