OctoPrint/OctoPrint #1863: from gcode calucalted size and time wrong

ManuelW77, 2017-04-12:
Hey there,

i was wondering about the much off Printtime calculated by Octo. Now i found, that Octo is not correct calculate the modelsize.

This i found in all my uploaded files:

> Modelgröße: 104.11mm × 52.53mm × 14719.80mm
> Modelgröße: 26.08mm × 26.07mm × 3834.90mm
> Modelgröße: 106.04mm × 101.93mm × 26679.82mm

I thing that is the height of the object, i haven't a printer that can print such dimensions :D
That causes completele off Printtimes. A object that takes about 2h will shown as 13h needet!

Here is a gcode file, created with S3D as example.
[Rader-hinten-Manu.zip](https://github.com/foosel/OctoPrint/files/915027/Rader-hinten-Manu.zip)

foosel, 2017-04-12:
> it dosn't matter

Oh for.... I wouldn't ask for it if it didn't matter. Please do not make assumptions about what matters and what doesn't for me to analyze potential bugs - having to run after this kind of information takes away time that could better be spent on fixing issues.

foosel, 2017-04-12:
Assuming you were running 1.3.2 (you didn't say) and assuming you were referring to the metadata displayed in the file list (you didn't say), the issue was that your GCODE specified a Z offset of -5mm which due to a bug was applied to all absolute positions, thus accumulating over the whole file.

I've pushed a fix to the `maintenance` branch (see above) that solves this and will soon also merge that to the `devel` branch. This will be part of the 1.3.3 release.

Marking this as solved since based on the information you provided I can only assume this was the issue you were facing.

foosel, 2017-05-31:
1.3.3 was just released.

closed 2017-05-31 by foosel

adrianmihalko, 2018-04-28:
I have same problem:

Model size: 152.48mm × 170.00mm × 26.40mm
Estimated total print time: 25 minutes

I resized my model in cura to 50%, so the size should be:

[image]

Estimated time in cura is 28 minutes. Octoprint shows me Print Time: 00:12:22
Print Time Left: 27 minutes (was 40, 2 minutes ago).

adrianmihalko, 2018-04-28:
The gcode zipped: [cat.zip](https://github.com/foosel/OctoPrint/files/1958161/cat.zip)