OctoPrint/OctoPrint #1865: invalid start/end gcode throws octoprint off

uwe999, 2017-04-12:
version 1.3.1:
I copied and pasted my start/end gcode from cura into octoprint.
There was a {travelspeed} variable in the end code which caused octoprint to kind of hang up at the end of the print. I had to restart octoprint in order to continue printing.
Maybe it should just skip invalid gcode and show a message on the UI?

Beste Gruesse
UWE

ntoff, 2017-04-13:
`Send: N431191 G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed}*81`

That "travel speed" is a cura variable which will be automatically filled out by cura when it generates the final gcode and probably caused your printer to freak out. Replace the variable with an actual number.

`G1 Z+0.5 E-5 X-20 Y-20 F2500`

foosel, 2017-04-13:
It would have helped if you'd provided the contents of your terminal tab or even better yet a serial log as requested in the template - I'm fairly sure that it's not OctoPrint that froze up but your printer, due to what @ntoff said (and what you already noticed yourself).

> Maybe it should just skip invalid gcode and show a message on the UI?

In order to be able to skip invalid gcode, it would first need to be defined what valid gcode is. And there is no such definition. While there is a standard for gcode, that's not the type of "gcode" that is used in 3D printing. There is some agreement as to how things are to be handled, but stuff still varies. So that is one problem here. Then there are things like in-band commands as supported by Smoothieware that don't even resemble GCODE at all but still are understood perfectly fine by the firmware. Additionally, the point at which OctoPrint would be able to decide if it's sending something invalid over the line after all line processing has been taken place and all registered plugins have had a chance to modify the line is so far down in the backend that it would simply not be possible to send a notification to the user. Also, OctoPrint can't know how much you've modified your firmware - maybe in your specific version `{travel_speed}` is a placeholder that will automatically be replaced with some internal firmware variable as needed? It's impossible to know it.

Which is why it boils down to this: you are the user, you know your printer and its firmware best. You decide what to sent to it and what not. OctoPrint will keep its hands off of things (unless you've told it not to through specific plugins or stuff like the temperature offset functionality) and that's it.

There's simply no way to "sanity check" your gcode scripts for you.

closed 2017-04-13 by foosel