OctoPrint/OctoPrint #1869: [Request] Material buttons to more quickly set temperatures

light24bulbs, 2017-04-14:
Right now it takes four clicks to set up temperatures for a material.  I have to use the drop down on the bed and the extruder to set the termpatures.  In the empty space under this dialog, I would like buttons that select the material and set that as the temperature.  It would be cool if they had a check mark or turned into a btn-success(green) when they were selected.  

Here is my terrible artists impression.  

[image]

ntoff, 2017-04-14:
You can add your own custom buttons to the "Controls" tab, maybe it'd be nice if that were extended to be able to add custom buttons to any tab? I know for sure I'd use that feature (I  currently have a few temperature buttons in the custom control area).

foosel, 2017-04-18:
This actually does sound like a duplicate of #216, so I suggest to follow that one instead and maybe add a thumbs up to it:

[image]

Closing this one because - as said - it's a duplicate :)

closed 2017-04-18 by foosel

light24bulbs, 2017-04-18:
Ah well...now both threads are closed, and neither has been resolved or answered

foosel, 2017-04-19:
I don't know what you are talking about, #216 is open.