OctoPrint/OctoPrint #1870: Weird behaviour with Marlin v1 and OctoPrint - Serial Communication not working properly - uncontrollable motor movements when switching to "control" tab in UI

HenryWhite24, 2017-04-14:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete any lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to repor tis still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the contribution
guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

I'm using OctoPrint for over a year now with a chinese Delta Printer based on RAMPS 1.4 and Marlin Firmware v1. A few weeks ago, I decided to update OctoPrint because I had not done this since at least 8 months.

After I updated, I couldn't use OctoPrint anymore because the serial communication shows some weird issues since then. Because it is so hard to explain, I made a youtube video which shows what happens when I try to connect through the printer **(see below)**

I thought there was something wrong with my OctoPi Installation, so I
- made a fresh installation of OctoPi (didn't resolve the issue)
- made a fresh installation of raspbian and installed octoprint manually (didn't resolve the issue either)

**The Problem only occurs with octoprint. When I connect to the printer, everything is okay. But when I switch to the "Control" tab, the problem occurs. Then I pull the serial cable because I don't want my RAMPS/ Arduino to be damaged.

With Windows and RepetierHost there is no problem in printer communication.**

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

Linux (Raspbian)

#### Printer model & used firmware incl. version

Chinese Delta Printer, Marlin v1

#### Browser and Version of Browser, Operating System running Browser

Has nothing to do with the browser, happens even when I set octoprint to "Connect automatically" and restart the system without leaving the web interface open in the browser.

#### Link to octoprint.log

https://pastebin.com/7QrcV1VG

#### Link to contents of terminal tab or serial.log

Doesn't show anything useful sadly.
https://pastebin.com/0dXSM4w9
Please note that the last 3 lines are caused by me pulling the serial cable.

#### Link to contents of Javascript console in the browser
JavaScript console? Do you mean the terminal window? It shows exactly the same as the serial.log above.

#### Screenshot(s) or video(s) showing the problem:
https://youtu.be/___K--hS_Qg
**Serial Communication starting @0:05
Switching to "control" tab in UI @0:19
Motor movements @ 0:26 (you need to crank up the volume so you can hear it)
**

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

Erutan409, 2017-04-14:
What about your syslog?

HenryWhite24, 2017-04-14:
@Erutan409 here's my syslog: https://paste.ee/p/QPIle

Erutan409, 2017-04-14:
```
Apr 14 09:59:58 raspberrypi kernel: [  116.294396] usb 1-1-port2: over-current change
Apr 14 09:59:58 raspberrypi kernel: [  116.509793] usb 1-1.4: USB disconnect, device number 4
Apr 14 09:59:58 raspberrypi kernel: [  116.510101] cdc_acm 1-1.4:1.0: failed to set dtr/rts
```

The last line looks VERY familiar to an issue I'm having while I tried switching my raspberry pi over to a switching power supply method.  Works fine if I'm using a traditional wall wart transformer.  But, something about the switching power supply was causing an issue where I'd send one move command from OctoPrint and then it crashes and I have to reboot the OS.

Is that what's happening for you?

Erutan409, 2017-04-14:
One thing you could try, if you haven't already, is adding `max_usb_current=1` to the bottom of your `/boot/config.txt` file.

HenryWhite24, 2017-04-14:
> The last line looks VERY familiar to an issue I'm having while I tried switching my raspberry pi over to a switching power supply method. Works fine if I'm using a traditional wall wart transformer. [...]
Is that what's happening for you?

@Erutan409 thanks! A new power supply did the trick. Now everything works as expected. Too bad I didn't come up with this idea myself before I went through the hassle with the all-manual installation :/

closed 2017-04-14 by HenryWhite24

Erutan409, 2017-04-15:
@HenryWhite24 updating the raspberry pi caused the problem to begin with, though?