OctoPrint/OctoPrint #1871: [request] temperature "off" button

ntoff, 2017-04-14:
@foosel Something I've wanted for a while now (and I think I saw someone else did too) is an off button on the main temperature page. It's just a couple of lines in temperature.jinja2, are you ok with this being in there? If so I'll make it a pull request.

@BillyBlaze Will adding an "off" button screw anything up for touchUI? A quick test suggests not but I've not tested it on the tiny screen of my phone yet.

(on PC)
[image]

(touchui on an android tablet)
[image]

ntoff, 2017-04-14:
https://github.com/foosel/OctoPrint/compare/devel...ntoff:dev/temperature-off-button

ntoff, 2017-05-03:
closing due to lack of interest / feedback

closed 2017-05-03 by ntoff

foosel, 2017-05-03:
Sorry, this slipped off my TODO list thanks to multiple factors. Finally got around to it today and pinged @BillyBlaze, and we were actually just discussing this on IRC. I'm still not completely sure about the actual need for this to be honest. It just feels a bit redundant and - a point that BillyBlaze brought up - somewhat dangerous since it might be easy to misclick. The current drop down approach avoids that and only adds one additional click, which might actually be a good thing when disabling heaters.

So I'm somewhat reluctant to do that for everyone. But it could maybe be done with a bit of jquery injection from a plugin?