OctoPrint/OctoPrint #1873: [Request] edit g-code in octoprint

userosos, 2017-04-16:
Hello everyone! I has new idea for octoptint. Will be good if i can edit (it will very god if it will full-featured text editor or just based functions) file with g-code command into octoprint. I searched and did not find in plugin repo.

toxuin, 2017-04-16:
Once this feature is implemented, it would help RepRapFirmware users to edit the config files on SD card as those are just g-code files.

JustAnother1, 2017-04-17:
As a tip to the implementer (If someone decides to implement this):

It probably makes sense to scan the file first and to then offer to edit the "sections"* of the G-Code file. The editor window would then only show the lines of that section. This will help with the amount of data that needs to be read from the file and send to the user. It would probably also be beneficial to the user as for most use case IMHO the user knows where he wants to edit. So this reduces the amount of scrolling the user has to do.

*What I mean with sections is: Stuff before the first G1 move could be the first "section" everything between two G1 moves that have a Z component is a layer section. And the stuff after the last G1 would be the final section. It could help to also parse comments. Cura for example adds comments for layer switches,..

userosos, 2017-04-17:
It can to look like in repieter-host after generate g-code i can press edit g-code.

Salandora, 2017-04-17:
Well Well I'm working on a GcodeEditor for quite some time now, but some other things have higher priority e.g. real life ^^, and another plugin ^^

My actual state is I can load files and get them displayed inside an editor... 
Next step (maybe in may, maybe june) is to save the file either by overwriting or modifing don't really know, overwriting would properbly be easier because all the functions needed are already there, and it would retrigger the analyser.
I just now notice that I startet the editor back in November 2015 what the ... where is all the time?
I guess I need to up the priority of this one now...

Also I think this is a duplicate of #1174 and #231 
Just incase I'm wrong: @foosel am I right? If yes shall we close this one?

ntoff, 2017-04-18:
What about an option to save or save as new file? I can see some people wanting to edit something and then save a new file so they have both versions to compare say a temperature change (which would be nice, not having to re-slice just to change that one line of gcode that controls the temperature).

Salandora, 2017-04-18:
@ntoff good idea, that would even be simple to implement, just a simple text box as filename, then a simple check if file exists, if yes show a dialog and ask if overwrite is okay, otherwise upload as new file.

userosos, 2017-04-18:
@Salandora Good! I think that need add templates. For example, i want change temperature in 10's layer and then i want change temperature in 50's layer and in 100 's layer i want on\off fan and other functions i want just do what i want and input value if need (for temperature, for example). It will be very good!

foosel, 2017-04-18:
@Salandora yes, this is very much a duplicate of #231. Closing for this reason, feel free to continue the discussion in #231 :)

closed 2017-04-18 by foosel

Salandora, 2017-04-18:
@userosos well If you find the specific code line no problem you can edit it in, I'm not sure if I will implement something like layer detection. 
A search function obviously, so if you know the Z height of the layer you can simply search for the specific G1 code and might end up at the right position...