OctoPrint/OctoPrint #1874: [Brainstorming] Allow to configure arbitary command for timelapse

Kunsi, 2017-04-16:
Currently, timelapses rely on having a Snapshot URL defined in Octoprint settings. It would be nice to be able to configure a command which gets executed instead of taking the snapshot inside Octoprint.

Points to consider:
- If both URL and Command are defined, which one should be used? (if both, in which order?)
- Should Octoprint handle the creation of the actual movie, if the command took pictures?
  - If yes, how does it know where to find the relevant pictures, and which pictures to use (if not all inside directory)?
  - If no, how to notify Octoprint there's a new finished timelapse available?
- **plugin only?** Haven't seen any functionality to have timelapses enabled, but prevent octoprint from taking pictures
- See also: #1854 

(Opening this as brainstorming, but may also be a Feature Request and/or a reminder to submit a pull request)

Kunsi, 2017-04-18:
Actually, it would be better to just turn Timelapse into a plugin completely, so there can be multiple Plugins available for specific use cases. Only thing inside core Octoprint which requires a Webcam is enabling keyboard controls when a webcam is defined - but that could also be enabled through the plugin (or, through a checkbox in settings).

foosel, 2017-04-19:
Let's try to separate the various components here:

  * something that takes snapshots - currently only an URL is possible here, and that gets used by a couple of plugins, the timelapse "recorder" and the (undocumented?) `/downloads/camera/current` endpoint, which all need to implement the actual image fetch themselves
  * something that coordinates the regular taking of snapshots and compiles a movie from the results - current timelapse implementation
  * something that embeds a webcam stream in the frontend - mjpg URL only right now, handled completely separately from the other two

First of all, I wouldn't bunch the third component together with the other two - completely separate usage and also separate config value.

Based on your requirements, I think the only thing that would need to be touched is actually the first bit here, basically creating some kind of central "SnapshotManager" or something like that, capable of fetching an image from wherever and returning it either as path and/or file handle (as requested). Adjusting the current timelapse implementation to utilize something like that would be fairly trivial. Where I see a problem is backwards compatibility to existing plugins, but that could be solved by having the standard implementation of the "SnapshotManager" simply still utilize the URL defined as snapshot URL and sub classes either using that as a base image to overlay with additional stuff or not. Adaptability through plugins could be achieved similarly to how we already allow injecting difference serial or printer implementations, e.g. [the `octoprint.printer.factory` hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-printer-factory).

Making the timelapse implementation in general adjustable through plugins (e.g. other types, other starting points and such) is a [larger task](https://github.com/foosel/OctoPrint/labels/epic%3Atimelapses_and_overlays) and I'd rather keep that separate from this one.

Upside of extracting the snapshot taking into its own (overridable) manager class is that it would be fairly easy to utilize that from a potential new implementation once that epic there gets tackled.