OctoPrint/OctoPrint #1875: [BUG] [Safari] Form Auto Fill messes up terminal input field

toxuin, 2017-04-16:
#### What were you doing?

After first command, Safari (may be other browsers with autofill enabled) tries to auto-fill the form based on previous input. This does put text in, but does NOT produce keyUp and keyDown events, where octoprint terminal form relies on.
This messes up the whole input field, disallowing me to input even a fresh string. Input can be done, but the command will never be sent to server.
After I manually add autocomplete="off" to the input field, everything works amazingly well.

#### What did you expect to happen and what happened instead?

I am not quite sure how this field internally works, but I would assume this should not rely on keyUp and keyDown as this would theoretically get messed up for copy-pasting too.

Input field in question is the one on the terminal page with id="terminal-command".

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

Server runs Raspbian, most likely irrelevant.

#### Printer model & used firmware incl. version

FIRMWARE_NAME: RepRapFirmware for RADDS FIRMWARE_VERSION: 1.17d ELECTRONICS: RADDS 1.5 FIRMWARE_DATE: 2017-01-28

#### Browser and Version of Browser, Operating System running Browser

MacOS 10.12.3 (16D32), Safari 10.0.3 (12602.4.8)

#### Link to octoprint.log

[link to gist.github.com](https://gist.github.com/toxuin/196b3d1592197c165d0e9d47942b42c1)

#### Link to contents of terminal tab or serial.log

[same link to gist.github.com](https://gist.github.com/toxuin/196b3d1592197c165d0e9d47942b42c1)

#### Link to contents of Javascript console in the browser

[same link to gist.github.com](https://gist.github.com/toxuin/196b3d1592197c165d0e9d47942b42c1)

#### Screenshot(s) or video(s) showing the problem:

Not applicable

I have read the FAQ.

foosel, 2017-04-19:
Could you make a gif or youtube video of this? I'm struggling to understand what exactly you are seeing, and I cannot reproduce anything that fits your description (or generally looking out of the ordinary).

KevMags, 2017-04-19:
Macs using Safari (10.1.1) have "Autofill>Other forms" enabled by default which insists on altering the gcodes entered in the terminal tab.  Adding a space to the end of newly entered text to make it a unique entry will allow the gcode to be sent/read properly.  I pulled the few hairs that I have left out trying to figure out why the M119 gcode entries were not being sent to the terminal.  Disabling  "Autofill>Other forms" in Safari preferences fixes the problem.

foosel, 2017-04-19:
Ah, that setting being at fault was a good hint. That's (still?) disabled by default on BrowserStack's Safari. Enabling it got me a reproduction, disabling auto completion on that input field indeed solved the issue. Pushed to `maintenance`, soon `devel`, will be in the 1.3.3 release.

Thanks!

BillyBlaze, 2017-05-03:
Just a small heads-up; webkit has started to ignore autocomplete="off" and other browser makers are following their policy. (See [comment](https://bugs.chromium.org/p/chromium/issues/detail?id=468153#c164))

So this fix might be a temporary solution.

foosel, 2017-05-03:
So setting this to `autocomplete="gcode-command"` should probably behave the same as "off"?

foosel, 2017-05-31:
1.3.3 was just released. Unsure what to do about this though, considering that the solution will soon be killed off by browser vendors :/

foosel, 2020-06-17:
Decided to close this after all. We've done what we can here, if browser vendors are being stupid about it, well...

closed 2020-06-17 by foosel