OctoPrint/OctoPrint #1877: Option to disable AUTO serial port

DaAwesomeP, 2017-04-18:
I run multiple OctoPrint instances on one machine. I have set udev rules so that no matter the order the printers are plugged in, they will always be `/dev/ttyPrusa`, `/dev/ttyXYZ`, etc. However, they are just symlinks to `/dev/ACM0`, `/dev/ACM1`, etc.

However, if one printer is unplugged or off when the system starts, then chaos ensues. For example:
 - instance 1: `/dev/ttyXYZ`
 - instance 2: `/dev/ttyPrusa`
 - but the XYZ printer isn't plugged in, so `/dev/ttyXYZ` doesn't exist and `/dev/ttyACM0` is mapped to `/dev/ttyPrusa`
 - instance 1 defaults to AUTO serial port instead of simply abandoning trying to connect to `/dev/ttyXYZ` and chooses `/dev/ttyACM0`
 - instance 2 errors because `/dev/ttyPrusa` is techinically locked (`/dev/ttyACM0`) by instance 1
 - instance 1 is connected to the wrong printer

An simple option to disable the auto-selection of serial ports would solve this issue.

foosel, 2017-04-19:
> instance 1 defaults to AUTO serial port instead of simply abandoning trying to connect to /dev/ttyXYZ and chooses /dev/ttyACM0

Are we talking "auto connect on startup" here? It is not clear to me in what connection scenario you are actually running into this problem, but if we are talking about auto connect on startup, that would be a bug - it shouldn't try to just happily fall back to AUTO if it can't see its preferred port during server startup.

Or are you by any chance running the PortLister plugin and are referring to its operation?

DaAwesomeP, 2017-04-19:
Yes, I'm referring to "auto connect on startup." It is defaulting to AUTO when the specified port isn't available. In preferences it shows AUTO as the specified port because the other port isn't currently available. I do not have the PortLister plugin installed.

foosel, 2017-04-20:
If the port you have specified as preferred port isn't available, it will show you AUTO in the connection drop down and the settings, simply because it then falls back on the built-in default - it has to select *something*. However, in such a case it should NOT try to do an auto connect on startup. I also just tested exactly that: configured a port as default and auto connect on startup, unplugged it, restarted the server. It did not attempt to auto connect, since it couldn't find the configured port it simply ignored the "auto connect on server startup" part.

Since I'm not entirely sure anymore right now what exactly you are seeing, please tell me which of the following two is it:

1. Your configured default serial port isn't available on server startup. OctoPrint still automatically connects, but to the wrong (not configured) serial port. No manual intervention of you at all.

2. Your configured default serial port isn't available on server startup. OctoPrint doesn't automatically connect to it, but pre-selects "AUTO" in the port dropdowns and when you plugin your printer in and then click "Connect" without noticing that it says "AUTO" there, you connect to the wrong printer.

DaAwesomeP, 2017-04-20:
I am experiencing the first scenario.
Part of instance 1 config:
```yaml
serial:
  additionalPorts:
  - /dev/ttyDavinci
  baudrate: 0
  maxCommunicationTimeouts: {}
  port: /dev/ttyDavinci
  timeout: {}
```
Part of instance 2 config:
```yaml
serial:
  additionalPorts:
  - /dev/ttyPrusa
  baudrate: 115200
  port: /dev/ttyPrusa
```
I didn't edit any of these manually; all were edited via the web UI.

foosel, 2017-04-20:
What version of OctoPrint is this?

DaAwesomeP, 2017-04-20:
Both are running v1.3.2 via master. When my current print finishes, I'll do a restart and get the latest logs with the issue.

DaAwesomeP, 2017-04-20:
I apologize. I restarted again and looked through the logs and found way more unrelated issues. One instance had messed up permissions and the other might have been corrupted somehow (I reinstalled). It works perfectly now! Thank you for your attention.

closed 2017-04-20 by DaAwesomeP

foosel, 2017-04-21:
Perfect, thanks for reporting back! :)

nophead, 2017-04-21:
A connected issue I have is that if I open OctoPrint in a browser when my machine is switched off the port goes to AUTO instead of /dev/ttyUSB0. When I switch it on the drop down doesn't allow me to set it back to /dev/ttyUSB0 unless I refresh the page. I would prefer it to always stay at /dev/ttyUSB0 and give an error if I press connect when the the port is not there. I don't ever want to see AUTO as I am not using auto detect.

foosel, 2017-04-21:
@nophead Install the [PortLister plugin](http://plugins.octoprint.org/plugins/portlister/), it will monitor the available connections and auto refresh the list for you when it sees something pop up.

nophead, 2017-04-21:
But why do I need that complication? I select the port I always one to use and save it, why can't that simply stay in the box all the time?

DaAwesomeP, 2017-04-23:
@foosel I agree, simply leaving the last selected port in the selector box but as a disabled option (grayed out) or labeled as "previously selected" would be informative.

foosel, 2017-04-24:
The box has always been a list of what is connected to your system, not what you might have connected to in the past (no tracking of that at all) or what you configured as default. 

Adding an option to a dropdown that we 100% know will immediately produce an error when chosen IMHO sounds like trouble. Marking it as "unavailable" and properly handling such unavailable options in the UI (not allowing the user to select it, marking it up in a special way, etc) could be done, but I still don't understand what that would provide you with. You would now see your port in the list with the info that it is not available. The same information is available now ("it's not in the list" means it is not available). You'd still need to refresh the list for it to become available and hence connectable.

So now we've spent quite some time adjusting the frontend to allow to display unavailable and unselectable options in this dropdown and gained basically nothing apart from a huge chance that after that change is introduced the next bunch of tickets that will be opened is "why does OctoPrint display a port there that is unavailable?". Sorry, but I really don't understand what would be the point of that.

What would make more sense is integrating the gist of that the PortLister plugin does - we can't do it like the plugin however since the way it solved this particular problem is not cross platform (monitoring a bunch of device files works on *nix, it doesn't work on Windows). We don't want to continuously poll for changes in the list when it's not necessary, so what would need to be done is detect when we are NOT connected and have the frontend poll the backend only then, plus maybe add a manual refresh button. And then find a way to have that not interfere with people's setups that already include the PortLister plugin (probably by having @markwal adjust the plugin to not do additional work for versions of OctoPrint that are adjusted accordingly). That could be done and would actually have a sensible cost-usefulness-ratio. 

I don't have time for that myself though in the foreseeable future as things are looking right now. So if you are in a hurry -> PR :)