OctoPrint/OctoPrint #1878: [Request] Add hook to instantiate different file type than PrintingGcodeFileInformation in comm.py

identify3d-joe, 2017-04-18:
Currently, in comm.py when a new file is opened, it must instantiate an instance of PrintingGcodeFileInformation. I would like a hook to instantiate my own class type from a plugin. This will give me a better way to control the file layer without making larger changes to the comm.py file.

An example plugin that I have written to test my changes is to put a gcode file in a zip file to save massive amounts of space. Then, when the file is loaded into the State bar, my class type handles the lower level io and provides the same information in real time/stream fashion as if it was uncompressed. 

Here is my suggestion (on or about line 821 of comm.py):
<pre>
file_factories = self._pluginManager.get_hooks("octoprint.comm.file.factory")
                        for name, factory in file_factories.items():
                                try:
                                        self._currentFile = factory(filename, offsets_callback=self.getOffsets, current_tool_callback=self.getCurrentTool)
                                        if self._currentFile is not None:
                                                self._logger.debug("Created file instance from factory {}".format(name))
                                                break
                                except:
                                        self._logger.exception("Error while creating file instance from factory {}".format(name))
                        else:
                                self._currentFile = PrintingGcodeFileInformation(filename, offsets_callback=self.getOffsets, current_tool_callback=self.getCurrentTool)
                        self._callback.on_comm_file_selected(filename, self._currentFile.getFilesize(), False)
</pre>
I can generate a pull request if access is granted. I have tested this code and it seems to work, but wanted you feedback on necessity and/or coding/naming style.

foosel, 2017-04-19:
In principle I like the idea very much. The thing is though that I plan to finally get around to revamping the comm layer with 1.4.0 (something that's burning a hole in my pocket for nearly three years now and got nuked on multiple occasions now thanks to external factors). 

That will probably also mean a lot of changes with regards to how jobs are modeled and handled in general, and I fear adding something like the above would cause some serious issues with backwards compatibility once those changes happen. I figure it would cause a lot of overhead trying to keep the mechanism working, which OctoPrint's versioning scheme would guarantee though.

I'd rather make the new layer allow this from the get go, but probably with an implementation more specific to the new architecture, than adding this now and risking huge road blocks due to having to maintain backwards compatibility then.

identify3d-joe, 2017-04-20:
I am a bit confused. I have seen at least one feature in your plugin infrastructure that carried a warning that the comm layer would change and you cautioned that the feature may change. I am not sure how this is different.

I don't want to tell you your business.....it's definitely your project and you can run it as you see fit. It just seems that you are making a sacrifice of a good feature to keep an image of a perfect solution......that will take 6 months to a year to release (and another 6 months to a year to become stable......the comm layer is always going to be tricky). I am really trying to be a good citizen and work on a mainline version and maintain a releasable plug-in, but this will force my work onto a fork and if we are taking that shortcut, it's much easier to just work off the fork instead of through the plug-in system. Again, it's your project and this isn't a threat.....it's just a reality of how my software team will react to these constraints for software that we are not trying to monetize and really only need for an internal prototype.

Thanks for your consideration and keep up the good work.

foosel, 2017-04-20:
> I am a bit confused. I have seen at least one feature in your plugin infrastructure that carried a warning that the comm layer would change and you cautioned that the feature may change.

Now I am confused because I am not aware of any such place in the docs. Would you happen to remember an example for that? I just went through the docs and couldn't find anything like that, but it's still morning ;)

> It just seems that you are making a sacrifice of a good feature to keep an image of a perfect solution

We are talking about a good feature I'd have to guarantee support for until 2.0 even though I already know that I'll revamp that particular part in 1.4.0. 

The problem here is simply that while you might be aware that things may change, and you might be happy to adapt your plugin when they do and might also have a solution in place to enforce people running older versions of your plugin to update and such, other people stumbling over such a hook might not. 

A plugin system whose documented interfaces keep changing all the time in a backwards incompatible manner is something I'd personally not want to develop for, so I can't really expect anyone else to either. Hence the decision early on to keep these things backwards compatible as much as possible (accidents do happen, but are to be avoided). Yes, this produces problems like this one here, but the alternative would have been to not offer plugin authors any kind of baseline they can depend on. And for this particular hook I already know that it would be hard to support down the road. 

I understand that worrying about things in half a year or more is kinda irrelevant for you and most people out there using this stuff, but I have to think of things like that and their consequences all the time and prevent introducing even more maintenance overhead since I'm already running at capacity as is. And trust me, a different approach would definitely be more fun for me as well, but hurt the project.

What I could offer you here is the following... you add the hook through a PR against `maintenance` (so for 1.3.x), document it such that it's *absolutely* clear that it is expected to vanish again in 1.4.0 without further notice or a backwards compatible alternative and that people using it do so at their own risk and should limit their plugins to only signal compatibility with OctoPrint<1.4.0 accordingly (incl. using ``__plugin_check__``). That means the only thing I have to guarantee is that it works with 1.3.x (and that should hopefully not add any maintenance overhad) and beyond that it's the plugin author's responsibility to change their implementation. It goes against OctoPrint's versioning scheme, but it should hopefully be clear enough that everyone understands that this is outside of the usual backwards compatibility guarantees and only there as a temporary solution.

identify3d-joe, 2017-06-09:
Just for the sake of documentation, here is a link to where you have put in a feature with a warning:

http://docs.octoprint.org/en/master/modules/printer.html

is_current_file

foosel, 2017-06-26:
`octoprint.printer.PrinterInterface.is_current_file` is completely unrelated to the comm layer (`octoprint.util.comm`) though O_O

With regards to the actual issue here, I made you an offer on how to achieve your goals for now by introducing your change for 1.3.x/`maintenance` with documentation marking it as something that will not be kept backward compatible. As I said I'd happily accept such a PR.

identify3d-joe, 2017-06-27:
Hi Gina,

   I was documenting a place where you warned that a layer would change so use at your own risk. I am not trying to be a pain here, more just as I was digging through your documentation, I came across where I was thinking of a spot where you had done it before. I apologize for the confusion.

  Unfortunately, after discussing with my team, we decided to just maintain our own fork and merge in your mainline changes as needed. After finding a few bugs, and coupled with the missing feature made it much simpler to work off of our fork where we could keep bug fixes and make it quicker to deploy. This does have the drawback of not publishing a generic version of our plugin externally, but that's okay from our point of view as we get more than enough testing on our printers. 

  Thanks for the offer and we appreciate the flexibility.