OctoPrint/OctoPrint #1879: Connection failure after M303 PID Tuning

pinthenet, 2017-04-20:
#### What were you doing?

Initiating M303 PID tuning from 'Terminal'- M303 E0 S210 C8 with M105 messages suppressed

#### What did you expect to happen and what happened instead?

Althought the PID tuning ran correctly with suggested Kp/Ki/Kd values, Octoprint lost connection shortly after comlpetion. The Printer LCD showed 'Waiting for user...'

Had to restart Octoprint & printer to reconnect - although maybe if I'd pressed the printer menu button it may have reconnected

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

Ubuntu 16.04.2 LTS

#### Printer model & used firmware incl. version

Velleman Vertex K8400 Marlin
Version M1-V1.4-H1 (http://vertex3dprinter.eu/support/downloads/vertex-downloads/)
Modified for dual E3D V6 extruders and GT2 pulleys & belts 

#### Browser and Version of Browser, Operating System running Browser

Not Applicable? (Chrome Version 57.0.2987.133 just in case)

#### Link to octoprint.log & serial.log

[Octoprint.zip](https://github.com/foosel/OctoPrint/files/942794/Octoprint.zip)

#### Link to contents of terminal tab or serial.log

see previous

#### Link to contents of Javascript console in the browser

not a UI issue

#### Screenshot(s) or video(s) showing the problem:

NA

I have read the FAQ.

Comment: I noticed a couple of M303 issues but they were closed and didn't seem to be the same. I realise it could be a Marlin bug. I have run M303 a few months ago but it was from an older Octoprint and on a Raspberry PI, I don't recall a connection issue.

foosel, 2017-04-20:
Yep, that's a firmware issue:

```
2017-04-20 12:30:48,488 - Send: M303 E0 S210 C8
2017-04-20 12:30:48,501 - Recv: PID Autotune start
2017-04-20 12:30:50,501 - Recv: ok T:20.34 @:127
2017-04-20 12:30:50,509 - Send: M105
2017-04-20 12:30:52,502 - Recv: ok T:20.78 @:127
2017-04-20 12:30:54,502 - Recv: ok T:22.37 @:127
2017-04-20 12:30:54,575 - Send: M105
2017-04-20 12:30:56,504 - Recv: ok T:24.97 @:127
2017-04-20 12:30:58,506 - Recv: ok T:28.53 @:127
2017-04-20 12:30:59,576 - Send: M105
2017-04-20 12:31:00,505 - Recv: ok T:32.54 @:127
2017-04-20 12:31:02,508 - Recv: ok T:36.10 @:127
2017-04-20 12:31:04,509 - Recv: ok T:40.08 @:127
...
```

The `M303` keeps sending one `ok` after another. `ok` means "hey, I'm done with a command, you may send the next one". OctoPrint tries that. And then gets completely confused when the printer refuses to acknowledge all the commands it did send as a response to the `ok`, plus it also probably overwhelms the serial buffer on Marlin's side in the process.

In a nutshell: The firmware must not send `ok T...` here during the ongoing PID tune. Only the `T` would be fine. One `ok` per sent command, not hundreds.

The problem are [these lines](https://gist.github.com/foosel/64b6a868c2b860a811d22f96cf01a5b7#file-temperature-cpp-L292-L295) in the firmware's `temperature.cpp` (at least according to the sources I found at the link you provided) - the `ok` in there is completely messing things up. From the looks of it, the problems should vanish if you remove the `ok` from what is written to serial there, recompile and flash the firmware. Appears to once again be a bug present in quite a large number of Marlin forks all over the net sadly, current versions appear to be fixed though as far as I can tell.

But based on the logs and the firmware behaviour, that is definitely not an OctoPrint issue and nothing I could even work around without completely breaking things for every other firmware.

pinthenet, 2017-04-20:
Thanks for your quick response. 
I thought the Recvs looked a bit odd during the PID.
Now I know it happens I can be prepared, it's not something I'll be doing too often (I hope). 

I'll report it on the Vertex forum. temperature.cpp wasn't changed as part of the E3D & belt/pulley mods.
Someone was working on bringing the Vertex version up to the latest Marlin revision, but not heard anything for a while. It seems to me that it's not doing the requested cycles either- I specified 8 but I only saw 6.

foosel, 2017-04-20:
That's the omnipresent problem with vendor forks of firmware, be it Marlin or something else... they rarely keep up with bug fixes that get merged upstream, and then you end up with issues like this all over the place :/

Anyhow, closing this if you don't mind since it's not an OctoPrint issue. I'll keep my fingers crossed that that's the only firmware bug you'll stumble over with this fork.

closed 2017-04-20 by foosel

pinthenet, 2017-04-20:
No problem to close. I've been using the printer for over 2 years and Octoprint for most of that so I hope this is the only bug....