OctoPrint/OctoPrint #1880: [Request] Ability to edit size and orientation of print.

HogensHero, 2017-04-22:
Hello,

I have gone thru all the settings and configurations as well as the documentation and FAQ and I cannot find where I am able to edit the *.gcode. I know that if I use Cura in windows I am able to edit the size and orientation of the print on the hotbed. It is my understanding that OctoPrint uses the Cura engine to print models so I was wondering if there could be a tab added with setting to adjust the model or if it could be added to the gcode viewer tab.

Thanks,

Russ

Printer: Anet A8
Computer: Raspberry PI 3
OS: OctoPi

ntoff, 2017-04-22:
Octoprint only uses the curaengine to *slice* your print, and turn it into gcode. It does that using an imported profile that you must create on your PC. If you want more control, check out the fully featured slicer plugin http://plugins.octoprint.org/plugins/slicer/ or the m33fio plugin (which does with with non m33 printers) http://plugins.octoprint.org/plugins/m33fio/ although it seems it's incompatible with the current version of octoprint so you'll have to wait for a fix. 

Once sliced and turned into gcode, there's no easy way to adjust the size and position of that model, that should be done in Cura's UI on your PC, then a new STL exported at the size you desire, and uploaded to the pi for slicing (assuming you're slicing using the raspberry pi).

foosel, 2017-05-02:
What @ntoff already said. Editing `.gcode` to change scaling/rotation/position on bed doesn't make sense. You need to do that before slicing your STL to GCODE.

And for that there is already a plugin available, plus there already is a request for it in #1333.

closed 2017-05-02 by foosel