OctoPrint/OctoPrint #1881: Pip command could not be found?

ChinookTx, 2017-04-22:
I'm having an odd issue where my OctoPrint does not seem to find python.  All my plugins appear to be non-functional.  Upon going to the Settings/Plugins Manager I get a warning that "The pip command could not be found or does not work correctly for this installation of OctoPrint".

I am told to look at the logs, but the Settings/Logs window is empty.

I had a quick glance at the /home/pi/.octoprint/logs, but eveything appears fine in there.

My pip in /home/pi/Oprint/pip reports "pip 8.1.1 from /home/pi/oprint/local/lib/python2.7/site-packages (python 2.7)"

My OctoPrint version is reported as "Version 1.3.2 (master branch)" in Settings/About OctoPrint.

This installation has been working fine for many months... I'm not sure what happened here...

READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

----

#### What were you doing?

[Please describe the steps to reproduce your issue. Be as specific as
possible here. The maintainers will need to reproduce your issue in
order to fix it and that is not possible if they don't know what you
did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen and what happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

[OctoPi, Linux, Windows, MacOS, something else? With version please.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure or reporting UI issues.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting any kind of communication issues between OctoPrint and your printer.
Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) or video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

ntoff, 2017-04-22:
Please compete the ticket properly, including all the information that is requested is necessary to track down bugs "everything looks fine" may not be enough.

>This installation has been working fine for many months... I'm not sure what happened here...

Well, the logs might shed some light but as they haven't been included...

Have you installed or changed anything? Do you always shut down your pi properly before unplugging it? Ever had to just hit the reset button?

ChinookTx, 2017-04-22:
Apologies for not including the log in the first place.  I have put the latest log file here: https://pastebin.com/raw/naAdnchq

Now strangely this morning after a reboot, it works (I did reboot about 20 times yesterday!!).  As for the previous questions, I don't remember changing anything in particular.  I do update plugins when I get notifications, but nothing more involved.  I do always shutdown the pi with a proper "shutdown -h now" or through the web interface.

Feel free to close the issue if you don't think it's worth investigating.  With no way to recreate it, it might be hard to debug unless something obvious jumps out in the logs.

foosel, 2017-05-02:
@ChinookTx my current hypothesis is that your frontend couldn't fetch the data from the server correctly, causing this message to be displayed even though everything was fine with your `pip` server side. This would also fit the quite large number of "access denied" responses logged in your `octoprint.log` file, especially those to `/api/plugin/pluginmanager` - no way to query the data from the server means error in the frontend:

```
2017-04-22 01:56:42,411 - tornado.access - WARNING - 403 GET /api/users (127.0.0.1) 10.00ms
2017-04-22 01:56:42,423 - tornado.access - WARNING - 403 GET /api/system/commands (127.0.0.1) 9.52ms
2017-04-22 01:56:42,435 - tornado.access - WARNING - 403 GET /plugin/announcements/channels (127.0.0.1) 9.21ms
2017-04-22 01:56:42,448 - tornado.access - WARNING - 403 GET /api/plugin/pluginmanager (127.0.0.1) 10.23ms
2017-04-22 01:56:45,350 - tornado.access - WARNING - 403 GET /api/languages (127.0.0.1) 10.66ms
2017-04-22 01:56:45,412 - tornado.access - WARNING - 403 GET /api/logs (127.0.0.1) 9.62ms
2017-04-22 01:56:56,195 - tornado.access - WARNING - 403 GET /api/languages (127.0.0.1) 11.71ms
2017-04-22 01:56:56,255 - tornado.access - WARNING - 403 GET /api/logs (127.0.0.1) 9.84ms
2017-04-22 01:58:34,386 - tornado.access - WARNING - 403 GET /api/languages (127.0.0.1) 19.99ms
2017-04-22 01:58:34,467 - tornado.access - WARNING - 403 GET /api/logs (127.0.0.1) 9.63ms
```

I could imagine that something went wonky with the login session, and that that fixed itself after a restart/reload.

Since you can no longer reproduce this, I'm closing this now, but if it happens again please request a reopen. We'll then have to try to find a way to reproduce it.

ChinookTx, 2017-05-02:
I've actually seen it again a couple of times, and it seems to be related to caching of some kind on the browser.  I have my password saved in my browser, and the "Remember me" box ticked.  One in a while it will get into this state, and even rebooting the Pi won't restore functionality.  Octoprint seems to think it's logged in since I get access to the settings page, etc.  

Long story short, I found that a single "logout" from the web interface, and logging back in restores everything...  I've seen it both with Chrome (latest) and Firefox (very old version), both on MacOS (2 seperate laptops).  

I don't know HOW to reproduce it, but I do know it will happen again!  If there are some things I can do next time it happens, I'll be happy to collect logs, etc.  Just let me know.

foosel, 2017-05-03:
It might be a very specific and very hard to reproduce race condition someone else saw a while ago related to the exact startup sequence when loading the web interface. I haven't yet found a proper solution for that. However, I *think* that only occurred on server restart and resolved itself after a reload, so it might not be the same issue after all. Then it happens again, it would be great if you could provide the JavaScript console output and also the response body of one of the failing requests. That might provide a hint as to why it's failing.

msross, 2017-06-23:
Logout from the web interface fixed the problem for me.

ChinookTx, 2017-06-26:
Another report of this happening:  https://github.com/guysoft/OctoPi/issues/372

foosel, 2017-06-29:
I might have found the reason for this (it's hard to say without being able to reproduce myself) and fixed it with the above commit.

The more people who test this the better though. I did my very best to keep things minimally invasive, but I did have to refactor some parts of the central web interface startup code and that might have as of now unforeseen side effect especially concerning third party plugins (even though I jumped through some hoops to avoid that). 

If so this should be fixed on `maintenance`, soon `devel` and be released with 1.3.5. Tentatively marking it as "solved" for now, but also as "needs testing".

foosel, 2017-06-30:
No, Maintenance RC are the release candidates only, and so far I haven't pushed any out for 1.3.5. So if you want to test right now you'd need to manually switch to the `maintenance` branch. Or just switch to Maintenance RC release channel and wait until the RC hits and report back then, that would also help a lot (as is reporting back on the RCs in general :))

foosel, 2017-10-17:
1.3.5 was released yesterday.

gdachs, 2017-12-09:
I get this `2017-12-09 16:12:25,938 - tornado.access - WARNING - 403 POST /api/system/commands/core/reboot (127.0.0.1) 18.24ms` with version 1.3.6rc3.
Logout and logging in again solves it. This doesn't happen from a remote session. It happens only local on my Odroid C2 where I start a chromium browser in fullscreen mode. As it happens only after some time idling, I assume it has to do with the session timeout that you mentioned before, but why is there no session timeout if I access remote?
[octoprint (5).log](https://github.com/foosel/OctoPrint/files/1545128/octoprint.5.log)

foosel, 2017-12-11:
Hmpf... I really had hoped this was fixed. Session timeout might be a good hint, I'll have to take a look at that.

jameswood, 2018-05-11:
I've had the same problem with Octoprint running on a Chip (as user "chip" instead of "pi" if that's at all relevant).

I have avoided the warnings by disabling Access Control and using HAProxy authentication instead. Happy to send logs etc if it will help, or if there's anything else I can do to help. Here's a snippet of what I was seeing with AC turned on:
```2018-05-11 10:02:35,254 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 40.36ms
2018-05-11 10:02:37,256 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 89.42ms
2018-05-11 10:02:37,349 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 76.45ms
2018-05-11 10:02:39,201 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 40.88ms
2018-05-11 10:02:39,252 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 39.87ms
2018-05-11 10:02:41,202 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 41.95ms
2018-05-11 10:02:41,257 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 43.90ms
2018-05-11 10:02:43,200 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 39.81ms
2018-05-11 10:02:43,241 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 31.90ms
2018-05-11 10:02:45,201 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 40.00ms
2018-05-11 10:02:45,256 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 43.87ms
2018-05-11 10:02:47,201 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 41.63ms
2018-05-11 10:02:47,252 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 39.83ms
2018-05-11 10:02:49,200 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 38.64ms
2018-05-11 10:02:49,255 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 43.37ms
2018-05-11 10:02:51,202 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 41.71ms
2018-05-11 10:02:51,254 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 40.74ms
2018-05-11 10:02:53,716 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 475.44ms
2018-05-11 10:02:53,746 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 23.12ms
2018-05-11 10:02:55,208 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 49.70ms
2018-05-11 10:02:55,250 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 31.80ms
2018-05-11 10:02:59,201 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 37.21ms
2018-05-11 10:02:59,250 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 33.41ms
2018-05-11 10:03:01,205 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 42.13ms
2018-05-11 10:03:01,257 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 40.73ms
2018-05-11 10:03:03,199 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 38.56ms
2018-05-11 10:03:03,255 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 44.09ms
2018-05-11 10:03:05,203 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 40.40ms
2018-05-11 10:03:05,257 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 41.29ms
2018-05-11 10:03:07,202 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 41.84ms
2018-05-11 10:03:07,258 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 43.81ms
2018-05-11 10:03:09,207 - tornado.access - WARNING - 403 GET /api/printer (127.0.0.1) 42.32ms
2018-05-11 10:03:09,261 - tornado.access - WARNING - 403 GET /api/job (127.0.0.1) 42.90ms```

Latest Octoprint, installed yesterday on a fresh OS for a new printer. Oh exciting times!

foosel, 2018-06-09:
I changed a bunch of things on the current `maintenance` branch that influence session handling, especially with regards to reverse proxies and ipv4 vs ipv6. I haven't yet pushed out a RC for 1.3.9, but once I do I'd ask anyone still running into this to give that a test and see if it fixes things.

foosel, 2018-07-25:
1.3.9 has just been released which contains a potential fix for this. Please test.

akutruff, 2018-08-04:
I'm running the docker version 1.3.9 on dockerhub, and I'm getting the same 403 spew.  The config file had accessControl  enable false at first. I set to true.  (could not login.) I set back to false.  Still get 403 spew.  Is this from accessing from a cached version of the front end from a previous version perhaps?  Chrome on PC accesses the site fine.  From my phone it will not load.

foosel, 2018-10-29:
Argh, for some reason I missed this reply... 

> Is this from accessing from a cached version of the front end from a previous version perhaps? Chrome on PC accesses the site fine. From my phone it will not load.

Possible, but hard to confirm. You could try clearing the cache in your phone, if the browser allows that. Are you seeing the 403 with any kind of endpoints, or only specific ones in the log?

lowlyocean, 2018-12-27:
I'm seeing similar errors with v 1.3.10. Have cleared cache in Chrome for the website, still seeing this after rebooting octoprint machine

`
13:32:35,551 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': <redacted>
13:32:39,845 - octoprint.server.util.sockjs - INFO - New connection from client: 24.191.117.30
13:32:40,337 - tornado.access - WARNING - 403 GET /api/files?recursive=true (162.158.62.39) 62.82ms
13:32:40,674 - tornado.access - WARNING - 403 GET /api/timelapse?unrendered=true (162.158.62.39) 33.79ms
13:32:40,707 - tornado.access - WARNING - 403 GET /api/connection (162.158.62.39) 27.05ms
13:32:40,748 - tornado.access - WARNING - 403 GET /api/plugin/pi_support (162.158.62.39) 30.52ms
13:32:40,788 - tornado.access - WARNING - 403 GET /api/printer/command/custom (162.158.62.39) 26.40ms
13:32:40,824 - tornado.access - WARNING - 403 GET /api/printerprofiles (162.158.62.39) 24.97ms
13:32:40,858 - tornado.access - WARNING - 403 GET /api/slicing (162.158.62.39) 23.50ms
13:32:40,892 - tornado.access - WARNING - 403 GET /api/settings (162.158.62.39) 23.81ms
`

foosel, 2019-03-19:
I still have not been able to reproduce that intentionally, but I'm suspecting some kind of caching issue. I've adjusted the caching headers in ceade46 and removed the need for the UI API Key in 17a7e50. We'll have to see if that helps people affected by this.

evanquinnalaska, 2019-05-10:
Same situation here:
OctoPrint 1.3.11rc3 running on OctoPi 0.15.1

two way's I access Octoprint:
Google Chrome | 74.0.3729.131 windows 10 OS
Octopod 2.1.1

[log.txt](https://github.com/foosel/OctoPrint/files/3168095/log.txt)

evanquinnalaska, 2019-05-10:
Rebooted and only used chrome Google Chrome | 74.0.3729.131 windows 10 OS

Just using auto cached creds and still showing errors. 

[log.txt](https://github.com/foosel/OctoPrint/files/3168113/log.txt)

evanquinnalaska, 2019-05-10:
Rebooted and only used chrome Google Chrome | 74.0.3729.131 windows 10 OS

Went Incognito to keep from using any cached information on the browser: Last log entry is sadly still Tornado.access
[log.txt](https://github.com/foosel/OctoPrint/files/3168133/log.txt)

evanquinnalaska, 2019-05-10:
Let me know if there is anything / way you would like me to change my behavior in testing to help debug :)

foosel, 2019-05-15:
This ticket will be the end of me...

foosel, 2019-06-04:
@evanquinnalaska I just got around to finally taking a close look at *all* your logs and noticed that you only get `403` responses on the `/sockjs/websocket` endpoint, which is served by the bundled SockJS Tornado dependency. Diving deeper down into that code I saw that a `403` with a response body of `Cross origin websockets not allowed` can be the result of a mismatching domain origin.

Which now begs the question: How are you accessing OctoPrint here? Do you maybe go through any further reverse proxies with different base URLs that might not be configured correctly? Any detail will be valuable here. Does the issue go away if you check "Allow Cross Origin Resource Sharing (CORS)" under Settings > API?

[image]

evanquinnalaska, 2019-06-04:
*Ways I access Octoprint:*
Slic3r (Prusa Slicer) to Upload / start prints.
Putty to SSH or Powershell to SSH into the Raspberry Pi 3b.
Chrome via windows 10 Computer.
Octopod via VPN on a Iphone.

*Solutions I use the API Key:*
*Octopod via VPN on a Iphone when away or Octopod via wifi when at the
house. *
*Required: *
*Name Generic Name*
*Hostname http://staticip <http://staticip> address*
*API Key: Octoprint's Generated API Key.*
*Username: Octoprint Username*
*Password: Octoprint Password*

Slic3r (Prusa Slicer) to Upload / start prints.
*Required:*
*Host Type: Octoprint*
*Host Name, Ip or URL: IP Address*
*API Key / password: Octoprint's Generated API Key.*

Both currently function.

Network is a simple home network on a single subnet between all devices
unless going through vpn. Vpn is using open vpn internally supported by the
Router.

Evan Quinn
Phone #: 1-907-299-9009

evanquinnalaska, 2019-06-04:
I will enable Cross Origin and let the log generate for a day and report
back to you.

Evan Quinn
Phone #: 1-907-299-9009

evanquinnalaska, 2019-06-04:
Cleared the logs, Enabled and Rebooted Pi.
Connect via Chrome. review log. No events.
Slice file via Prusa Slic3r send to octoprint.
Connect Via Octopod over VPN: Tornado.access Warning generated.

Evan Quinn
Phone #: 1-907-299-9009

evanquinnalaska, 2019-06-04:
Attached Log.

Evan Quinn
Phone #: 1-907-299-9009

foosel, 2019-06-04:
That sounds like either OctoPod is doing something weird then here, or the VPN is at fault. Not seeing and 403 in the log though 🤔

@gdombiak can you maybe elaborate on how OctoPod connects to OctoPrint's websocket? Does it set any kind of Origin headers that might explain this?

foosel, 2019-06-04:
In any case, I'm starting to think this is a different problem since it's as far as I understand only the `/sockjs/websocket` endpoint for which 403 responses get generated, which is handled completely different than the API endpoints that this ticket is about, so I might split this off into its own ticket at one point. Just as a heads-up ;)

evanquinnalaska, 2019-06-04:
Sure thing, I will also test once onsite on the wifi to validate it doesn't
generate 403 errors in the log.

Evan Quinn
Phone #: 1-907-299-9009

gdombiak, 2019-06-04:
Hi @foosel, after OctoPrint 1.3.11 was released I had to push a new version of OctoPod to include the _Origin_ header since it was giving a 403 error under certain circumstances. I was able to connect fine from LAN but noticed the 403 when coming from the Internet through a reverse proxy. Enabling CORS was a quick hack that was reported to work. I quickly pushed a new release of OctoPod to include the missing header.

evanquinnalaska, 2019-06-04:
Just for information sake: I am running OctoPod 2.3 on iOS 12.2.

Evan Quinn
Phone #: 1-907-299-9009

gdombiak, 2019-06-05:
@foosel, I got confused. I just checked the code and the reported 403 error with websockets got resolved when I added a _Host_ header in the initial HTTP request for establishing a websocket connection. 

I can try adding the _Origin_ header to match the header of the specified hostname. Would this be the reason for this 403? I haven't seen other reports about the 403 since I fixed https://github.com/gdombiak/OctoPod/issues/176.

Thanks,
Gaston

gdombiak, 2019-06-05:
I did some tests and I can confirm that when I remove the _Host_ header the 403 error happens again. Adding _Origin_ while keeping _Host_ out still produces the 403 error. However this is not @evanquinnalaska case since he is running OctoPod 2.3. 

The bad news is that I cannot reproduce the 403 error reported here when _Origin_ is missing and _Host_ is present. Let me know if there is more information that might help me reproduce this so we can test any potential fix. BTW, my tests are with CORS disabled.

foosel, 2019-06-05:
I've created #3193 for this since it really feels completely different than the original issue reported here, considering that it only happens with a specific third party client so far. Let's move further discussion into that ticket.

foosel, 2019-11-05:
Tentatively calling this finally fixed since I haven't heard anything about it anymore over the course of 1.3.11 being stable. Let's hope for the best.

closed 2019-11-05 by foosel