OctoPrint/OctoPrint #1887: OctoPrint does not connect to Blocks Zero

kodacy, 2017-04-26:
Installed on: Raspberry Pi 3 B 1.2

Printers page and information:
http://www.blockstec.com/zero.html
http://www.blockstec.com/zero.html

----

#### What were you doing?

Trying to conect OctoPrint 1.3.2 to my Blocks Zero Printer

When I press connect it goes from Off-line to Connecting and then back to Oflfine

Printers page and information:
http://www.blockstec.com/zero.html
http://www.blockstec.com/zero.html

#### What did you expect to happen and what happened instead?

Establish a connection has it happens with Printrun out of the box.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Blocks Zero firmware 1.0

#### Browser and Version of Browser, Operating System running Browser

windows 10 64-bit, Chorme

#### Link to octoprint.log
https://www.pastiebin.com/59011aa4cd60e

#### Link to contents of terminal tab or serial.log
OctoPrint Connection log:
https://www.pastiebin.com/5900f7cadfeaf

Printrun Connection log and some some send tests:
https://www.pastiebin.com/590100a907bc8

kodacy, 2017-04-27:
I think it is complete now any way to remove a incomplete ticket label?

foosel, 2017-04-27:
It would have done that automatically if you'd copied the *full* ticket template ;) But everything important is the, marking ok manually. 

Your issue is your firmware, it never sends an ok back for either of the attempts to talk to out (M110). Of it isn't already, try enabling to wait for start on connect under Settings > Features. That might do it.

kodacy, 2017-04-27:
Thank you for your feedback

did has you advised with no success 
http://prntscr.com/f1hhs6

still getting back:
There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor

foosel, 2017-04-27:
`serial.log`/terminal output still looking exactly the same? There should have been at least some change in there (if not, your changes weren't properly applied), please provide output how it looks now. 

To re-iterate, your printer is not responding back here properly. It should acknowledge at least *one* of the sent `M110` commands with an `ok`. Since it doesn't OctoPrint can't complete its initial connection handshake and gives up. 

Sometimes the reason is that whatever gets sent to the printer while it's still "starting up" gets eaten, in which case the "wait for start" option can help to get things going. But if OctoPrint is now not sending anything to your printer until it sees its `start` output and stuff still doesn't work, it is something else *on the side of your printer*. It might be that for whatever reason your printer doesn't understand that command (which would be seriously weird, but who knows). If so you could also substituting it with an `M105` (for example) under Settings > Serial Connection > Advanced Options > "Hello" command, see how that pans out.

The problem here is that unless your printer replies to OctoPrint's first command with an `ok`, OctoPrint can't be sure if it's talking to a printer or a coffee pot. So it assumes the worst if no well formed reply is received back within the timeout timeframe and gives up.

This:

    Recv: echo:Marlin1.0.0

indicates an old-ish Marlin fork. The problem with those is that they can be buggy like mad and rarely get fixed by the vendors with fixes from upstream. Or it reports "Marlin" there but is actually something self-built that behaves completely differently. Hard to say, never heard of that printer before either.

----

As a side-note, this from your `octoprint.log`:

```
2017-04-26 19:37:05,197 - tornado.access - WARNING - 404 GET /common/info.cgi (127.0.0.1) 11.78ms
2017-04-26 19:37:05,221 - tornado.access - WARNING - 404 GET /common/info.cgi (127.0.0.1) 13.18ms
2017-04-26 19:37:05,254 - tornado.access - WARNING - 404 GET /common/info.cgi (127.0.0.1) 11.70ms
2017-04-26 19:37:05,291 - tornado.access - WARNING - 404 GET /common/info.cgi (127.0.0.1) 11.81ms
2017-04-26 19:37:05,315 - tornado.access - WARNING - 404 GET /check_user.cgi?user=test&pwd=test (127.0.0.1) 11.96ms
2017-04-26 19:37:05,354 - tornado.access - WARNING - 404 GET /top.htm (127.0.0.1) 11.67ms
2017-04-26 19:37:05,382 - tornado.access - WARNING - 404 GET /top.htm (127.0.0.1) 13.29ms
2017-04-26 19:37:05,409 - tornado.access - WARNING - 404 GET /cgi-bin/CGIProxy.fcgi (127.0.0.1) 11.72ms
2017-04-26 19:37:05,430 - tornado.access - WARNING - 404 GET /index.html (127.0.0.1) 11.57ms
2017-04-26 19:37:05,540 - tornado.access - WARNING - 404 GET /index.html (127.0.0.1) 11.74ms
2017-04-26 19:37:05,562 - tornado.access - WARNING - 404 GET /index.html (127.0.0.1) 11.73ms
2017-04-26 19:37:05,671 - tornado.access - WARNING - 404 GET /SDK/activateStatus (127.0.0.1) 13.15ms
2017-04-26 19:37:05,694 - tornado.access - WARNING - 404 GET /SDK/activateStatus (127.0.0.1) 11.93ms
2017-04-26 19:37:05,809 - tornado.access - WARNING - 404 GET /cgi-bin/get_status.cgi (127.0.0.1) 11.62ms
2017-04-26 19:37:05,832 - tornado.access - WARNING - 404 GET /get_status.cgi (127.0.0.1) 11.71ms
2017-04-26 19:37:05,942 - tornado.access - WARNING - 404 GET /get_status.cgi (127.0.0.1) 11.93ms
2017-04-26 19:37:05,968 - tornado.access - WARNING - 404 GET /CgiTagMenu?page=Top&Language=0 (127.0.0.1) 13.50ms
2017-04-26 19:37:06,078 - tornado.access - WARNING - 404 GET /CgiTagMenu?page=Top&Language=0 (127.0.0.1) 11.95ms
2017-04-26 19:37:06,101 - tornado.access - WARNING - 404 GET /CgiTagMenu?page=Top&Language=0 (127.0.0.1) 12.10ms
2017-04-26 19:37:06,219 - tornado.access - WARNING - 404 GET /util/query.cgi (127.0.0.1) 11.55ms
2017-04-26 19:37:06,241 - tornado.access - WARNING - 404 GET /util/query.cgi (127.0.0.1) 11.93ms
2017-04-26 19:37:06,267 - tornado.access - WARNING - 404 GET /util/query.cgi (127.0.0.1) 13.32ms
2017-04-26 19:37:06,290 - tornado.access - WARNING - 404 GET /util/query.cgi (127.0.0.1) 11.92ms
2017-04-26 19:37:06,334 - tornado.access - WARNING - 404 GET /util/query.cgi (127.0.0.1) 11.86ms
2017-04-26 19:37:06,359 - tornado.access - WARNING - 404 GET /util/query.cgi (127.0.0.1) 11.67ms
```

looks *very* fishy. If you have a port forward running to your OctoPrint instance, making it reachable from the internet, I'd strongly suggest to add some additional access control in front or using a VPN, because that looks like some automated attempts at owning your machine.

Additionally, there's a random restart of the server in that log too (ALWAYS shut down your Pi properly, never just yank the power) plus some date/time weirdness:

```
2017-04-26 19:39:59,393 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/, --user flag needed => no, virtual env => yes
2017-04-26 19:40:01,488 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-04-25 12:17:06,878 - octoprint.server - INFO - ******************************************************************************
2017-04-25 12:17:06,880 - octoprint.server - INFO - Starting OctoPrint 1.3.2 (master branch)
2017-04-25 12:17:06,880 - octoprint.server - INFO - ******************************************************************************
[...]
2017-04-25 12:17:16,374 - octoprint.util.pip - INFO - Version of pip is 8.1.1
2017-04-26 21:57:31,866 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/, --user flag needed => no, virtual env => yes
2017-04-26 21:57:32,251 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
```

This is probably unrelated but still something that looks off.

ntoff, 2017-04-27:
Installed their firmware on an old arduino board and changing the "hello" command in Octoprint gets the firmware to connect, however it locks up on sending M110 N0

    Recv: echo:SD init fail
    Recv: ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 @:0 B@:0
    Changing monitoring state from 'Connecting' to 'Operational'
    Send: N0 M110 N0*125
    Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
    Send: M105
    Recv: ok T:20.8 /0.0 B:0.0 /0.0 T0:20.8 /0.0 @:0 B@:0
    Send: M105
    Recv: ok T:20.5 /0.0 B:0.0 /0.0 T0:20.5 /0.0 @:0 B@:0

It looks like pronterface doesn't bother with M110 until you  start a print, instead opting for M105 as their hello command, assuming if it gets back a temperature reading then it's talking to a printer maybe? (Just going by pronterface's log, I don't know if that actually is what it's doing).

	Connecting...
    SENT: M105
    RECV: start
    Printer is now online.
    RECV: echo:Marlin1.0.0
    RECV: echo: Last Updated: Apr 28 2017 01:53:16 | Author: (Blocks zero)
    RECV: Compiled: Apr 28 2017
    RECV: echo: Free Memory: 3634  PlannerBufferBytes: 1232
    RECV: echo:Hardcoded Default Settings Loaded
    RECV: echo:Steps per unit:
    RECV: echo:  M92 X80.00 Y80.00 Z4000.00 E97.00
    RECV: echo:Maximum feedrates (mm/s):
    RECV: echo:  M203 X250.00 Y250.00 Z3.00 E50.00
    RECV: echo:Maximum Acceleration (mm/s2):
    RECV: echo:  M201 X3000 Y3000 Z100 E10000
    RECV: echo:Acceleration: S=acceleration, T=retract acceleration
    RECV: echo:  M204 S2000.00 T1000.00
    RECV: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
    RECV: echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.40 E5.00
    RECV: echo:Home offset (mm):
    RECV: echo:  M206 X0.00 Y0.00 Z0.00
    RECV: echo:PID settings:
    RECV: echo:   M301 P22.20 I1.08 D114.00
    RECV: echo:SD init fail
    SENT: M105
    RECV: ok T:20.9 /0.0 B:0.0 /0.0 T0:20.9 /0.0 @:0 B@:0
    SENT: M105
    RECV: ok T:20.9 /0.0 B:0.0 /0.0 T0:20.9 /0.0 @:0 B@:0
    Disconnected.

One main difference I see is Octoprint sends `Send: N0 M110 N0*125` upon starting while pronterface sends `SENT: N-1 M110*15` once you hit that "print" button.

kodacy, 2017-04-27:
Again thank you for your help!

1st here is a fresh log with the the Wait for start on connect applied and then with Wait for start on connect applied flowed by the change in the hello comand:
https://www.pastiebin.com/59022c31bb18e

after a few long minutes the printer comunicante a bit more and moved the z axis back. but still no control over it as you can see on the log:
https://www.pastiebin.com/590232afd6652
http://prntscr.com/f1k1gq

I noticed the Merlin version was a bit old but the printer is quite new...

my ocrtoprint RPi 3 is not yet configured for remote ascese and i have no port forwarding for it, and it is a fresh image from your site (maybre do to une of the installed pulg ins)??? http://prntscr.com/f1jzat

Random restarts my foul (try to prevent cable yanks from now on)

If you find it useful i can do a fresh reinstall of the Octoprint image. with the main version or even the ne 14 beta.

foosel, 2017-04-27:
@ntoff thanks for that test! You are right, pronterface uses `M105` as hello command. The problem with that is that there are a couple of repetier boards out there that don't reset on serial connect, leading to errors when you try to connect to them with a single `M105` (requiring line numbers + checksums & no way to figure out correct line number by OctoPrint = error on sending). Thus OctoPrint uses `M110` as hello command. And it uses `M110 N0` explicitly, which gets rewritten to `N0 M110 N0*125` because there are also some older Marlin (?) dialects that still had a bug leading to weird line number parsing from the M110 command if no parameter was provided, and some other firmwares requiring M110 to be sent with line number and checksum always attached, and of course repetier which (in default configuration) requires line number and checksum with every command. Hence two `N0` and a checksum, where `M110` on its own would be so much more readable. 3d printer firmware. Lovely.

Anyhow, re the problem at hand, I took a quick peek at the firmware sources and from the looks of it it does support M110 but simply does not produce any output for it. Yay, awesome.

@kodacy Let's try something. Place the contents of [this](https://gist.github.com/foosel/936a05332fe517ed714f3fc557467083) as `fix_blockzero_stupid.py` in your `~/.octoprint/plugins` folder on your OctoPi installation. To do this via an SSH connection to your Pi this boils down to these commands:

    cd ~/.octoprint/plugins
    wget -Ofix_blockzero_stupid.py https://goo.gl/9PcgQk

Then restart OctoPrint and attempt to connect again. What this plugin does is basically just "simulating" the `ok` for the M110 that your printer's firmware refuses to send. That is a bit ewww because if something should go wrong with the M110 there's no way to detect this, but that's the case now as well, so nothing lost and this actually might get things going.

> [weird access lines in log]

If you don't have port forwarding enabled, that's a relief. Maybe something on your LAN scanning attached machines? It claims to come from 127.0.0.1 (so the Pi itself), but that could also be shadowed by forwards. Definitely looks odd and is not something you'd usually see with a fresh image.

> [reflash image]

As I said, I don't think the log stuff is related to your connection problems, which are rather caused by the firmware your printer vendor provides being broken. Let's try to work around that first.

kodacy, 2017-05-02:
@foosel and @ntoff Aguin thank you for your suport.

Did the process has you described:
________________________________
pi@octopi:~ $ cd ~/.octoprint/plugins
pi@octopi:~/.octoprint/plugins $ wget -Ofix_blockzero_stupid.py https://goo.gl/9PcgQk
--2017-05-02 10:02:43--  https://goo.gl/9PcgQk
Resolving goo.gl (goo.gl)... 172.217.23.46, 172.217.23.46, 172.217.23.46, ...
Connecting to goo.gl (goo.gl)|172.217.23.46|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://gist.githubusercontent.com/foosel/936a05332fe517ed714f3fc557467083/raw/55160b5d97086ce3c76fe084eb7581cffdd081cc/fix_blockzero_stupid.py [following]
--2017-05-02 10:02:43--  https://gist.githubusercontent.com/foosel/936a05332fe517ed714f3fc557467083/raw/55160b5d97086ce3c76fe084eb7581cffdd081cc/fix_blockzero_stupid.py
Resolving gist.githubusercontent.com (gist.githubusercontent.com)... 151.101.16.133
Connecting to gist.githubusercontent.com (gist.githubusercontent.com)|151.101.16.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 402 [text/plain]
Saving to: ‘fix_blockzero_stupid.py’

fix_blockzero_stupi 100%[=====================>]     402  --.-KB/s   in 0s

2017-05-02 10:02:44 (1.70 MB/s) - ‘fix_blockzero_stupid.py’ saved [402/402]

pi@octopi:~/.octoprint/plugins $ sudo reboot

Broadcast message from pi@octopi on pts/0 (Tue 2017-05-02 10:03:14 UTC):

The system is going down for reboot NOW!
______________________

But I do not see any difference in the printers behavior:
https://www.pastiebin.com/59085b16a75a3

Never the less, since this problem is mostly do to and outdated version of the merlin on the Blocks Zero Printer i've been allready confirmed that Blockstec are soon releasing a new firmware based on a newer stable version of merlin (1.0x ???), that should solve and the comuinication issues.

I'll beta test it and will shortly feddback to you.

foosel, 2017-05-02:
Oh, but there is a change in behaviour. It now completes the handshake, but then the `M115` and `M20` commands ALSO don't get properly acknowledged by the firmware with an `ok`. Something in this firmware appears to be really broken. So I got curious and took another look into the code and look at that:

``` c
void ClearToSend()
{
  /*previous_millis_cmd = millis();
  #ifdef SDSUPPORT
  if(fromsd[bufindr])
    return;
  #endif //SDSUPPORT
  SERIAL_PROTOCOLLNPGM(MSG_OK);*/
}
```

Sending `ok` is generally commented out. I have no idea what your printer vendor thought when they did that, but this firmware is completely and utterly broken in that state, in a way that is not recoverable. The firmware needs to be fixed, and I sincerely hope that this problem above is fixed in whatever they promised you.

Really... just wow. Not your fault of course. But I'm speechless right now.

Anyhow, closing this as it's an issue caused by an utterly broken firmware build, not OctoPrint. I strongly suggest to complain to your vendor, that they shipped this in the first place is a serious WTF.

closed 2017-05-02 by foosel

kodacy, 2017-05-04:
After testing the latest firmware it seems works out of the box with Octoprint. 👍 everyone should update to Blocks_zero_1.2.hex still testing and feedback on any problems.