OctoPrint/OctoPrint #1889: [Brainstorming] Plugin: Periodically cache last point sent in GCode and provide a recover print option.

SlightlyCyborg, 2017-04-27:
I fuxx0red my 37 hour print while paused on Repetier server to fix a belt jumping issue. As I was manually working on the printer, I unplugged it. Repetier server noticed the disconnect and stopped the print. FML

So, I am switching host software. Octoprint seems good and I like the community support I see. I plan on implementing a recovery feature, so I can literally unplug and move my printer if I want to mid print and start back up from where I left off.

To me it seems like this feature will need a different implementation for each slicer/firmware combo, I don't really know how other combos implement their movement. With slic3r and marlin, the gcode uses absolute positioning, and periodically issues a G92 command to "reset" the extruder positioning (I call this pseudo relative extruder positioning)

1) Manually home the printer, heat up the extruders.
2) Search for the last G1 command sent for each axis value
3) Send a G1 command with each last known axis values
3) Issue a `G92 Ex` command, where x represents the extruder motor position at the last sent G1 command. This will set up the extruder to work correctly with the next psuedo relative extruder position issued.
3) Start running Gcode from the last G1 command sent.

SlightlyCyborg, 2017-04-27:
relevant comment from the reddit post I made about this.

```
...one of the issues (maybe small, maybe large) is that the host has absolutely NO idea what the printer last did. The host software knows what it last sent to the printer, but not how far through the print buffer the printer is or whether the last sent command was ever executed (or in the case of a pulled USB cable, whether the last line was even seen by the printer).

So while it's certainly possible to re-start somewhere near where it left off, without some sort of feedback from the printer as to what line it last executed then there's no way to be sure you're starting exactly where you left off, which might result in a missing layer or two or even more depending on the size of the feature, the size of your printers buffer, and how many moves it was in to that buffer.

One exception to that rule might be repetier firmware which has an option to echo back commands when executed, that combined with some fancy host functionality might provide a way to know what the printer is up to but I'm not sure that works how I think it works, I've never actually tried it.
```

SlightlyCyborg, 2017-04-27:
One guy made a really relevant point "most of the prints that I've been doing fill up the entire print bed. When restarting a print from a certain place, by default, my machine will home the printhead, which requires it to tap down onto the build plate. My prints are so big that I can't do that , because the print keeps the printhead from being able to touch down."

So, I will just have to assume that the z axis is in the correct spot and home the x and y axises. The z axis really shouldn't move much on my printer since it uses a screwgear, while the x and y use belts. I am not sure what other printers use.

SlightlyCyborg, 2017-04-27:
So my new strategy is to home the x and y axises but assume that the Z axis is in the right position. If it isn't, I can manually move the z axis to touch the print that has been left on the bed. I could then use a G92 to set an offset which will properly configure the z axis.

ntoff, 2017-04-27:
Take a typical CNC approach to it. 

"Move the dumb machine some place, press some buttons to tell it where it is, hit start."

Home x and y, and just use G92 to tell it where Z is (or fetch it from the cache if you're looking to remember the last position and store it in some kind of non volatile memory).

I'd be kind of weary about restarting it though, especially on small objects, once a print bed has cooled things tend to pop loose, I'd be worried about one of them being knocked over and summoning the giant spaghetti monster.

While resuming a print would be a welcome addition to any 3d printers arsenal, I think it would have to come with quite a few warnings and caveats. Things like don't feed your printer after midnight, and don't get it wet... no, wait, wrong warning label... Things like "if the print has been stopped long enough that the bed and print have cooled, resuming may fail if the part-to-bed bond has been broken" and "there may still be a visible seam where the job has been resumed".

I think in my mind I'd rather just remove everything and print the last remaining part by itself and glue it on later.

trgtylcnky, 2017-05-03:
This feature could be also used to print remaining part as a new print, then parts are joined with glue maybe.

Since OctoPrint saves the file position if a communication error occurs, we can use this feature. When I look at the code, I see this is called by _recordFilePosition() function [here](https://github.com/foosel/OctoPrint/blob/0a69dbeddb301d5a32827a3f0d561f875df24234/src/octoprint/util/comm.py#L431)
We can hook to gcode sent event in our plugin then call this function.
Then, we also see how to start new print from a certain location [here](https://github.com/foosel/OctoPrint-PrintRecoveryPoc/blob/master/octoprint_printrecoverypoc/__init__.py#L77) 

I am also thinking about using a battery in order to save enough time to pause the print, finish buffered commands, then move the tool to an appropriate place to prevent oozing on the print after detecting the power outage.

foosel, 2017-05-03:
`_recordFilePosition` gets called on disconnect and on manual cancel when a print is ongoing. I'd strongly suggest to not call this on every sent GCODE line manually, because that will probably add so much processing overhead that you can pretty much forget about streaming data to the printer in a timely manner.

However you could record things you want to persist in memory, on error write them to your plugin data folder, and use that PLUS the print recovery data persisted by OctoPrint to process things further.

For reference see also [the README of OctoPrint-PrintRecoveryPOC](https://github.com/foosel/OctoPrint-PrintRecoveryPoc) and its source code in general.

SlightlyCyborg, 2017-05-03:
I used standard plugin hooks to hook into gcode sent. I acquire a lock,
increment a gcode counter, release the lock and run as normal.

I have a thread that acquires the same lock, grabs the current gcode line
number, releases the lock, writes it to a file saved in the plugin data,
and then delays for 5 seconds.

foosel, 2017-05-03:
The line number is completely irrelevant. It is arbitrary, can at any time be reset via M110 and the same line from the same file when printed can have differing numbers due to sightly different timing of eg temperature polls or manual commands sent via the terminal tab. Leave the line number tracking to OctoPrint, consider it part of the serial communication, *not* the print job (because it actually is).

The only things interesting for print recovery are the *byte position* of the file that is being printed and things like fan and temperature settings, position information (tricky to get because the only authority on that is the firmware and on disconnect that is gone), selected tool, extruded length (see position for issues with that) and maybe things like extrusion and feedrate multiplier.

SlightlyCyborg, 2017-05-03:
My implementation can be seen here: https://github.com/SlightlyCyborg/OctoPrint-Rescue/blob/master/__init__.py#L30

I am still working on putting to together the correct GCODE, saving it to uploads folder, and then selecting it.

SlightlyCyborg, 2017-05-03:
> byte position
 Yes, using byte position seems better than my current implementation. 

Grabbing self._printer._comm._currentFile.tell() should be the correct number I want to save?

MiggyMan, 2017-05-03:
As a note (more something to keep in mind), if your recording this back to nv media rather than memory and this is run on something like a pi booting from a sd card/usb stick you may well reduce the life of the media with by writing too it aggressively.

foosel, 2017-05-03:
`_currentFile` is not a file object. `tell` is not available on it, `getFilepos()` is. Take a look at the implementation of `_recordFilePosition` and `getFilePosition` which it calls. But in general be aware that anything that is not documented on docs.octoprint.org does not guarantee backwards compatibility. Especially not the current comm layer implementation (so if you are calling `self._printer._comm.*`, you are on thin ice already). It should be taken with a pinch of salt and avoided where possible.

I'd rather suggest to use what is already there and depend on that (built-in recovery data), enhance it with whatever else you want to track and be done. The current recovery data persistence should cover all bases that you possibly can cover  from OctoPrint or a plugin for that matter (cancelling due to a "oh shoot, need to do X first" situation, catastrophic communication error).

The hard problems here are actually not getting the position in the file out of the system (because as said, it already does save that for you, you just have to access that data) but in fact things like position tracking of all four axes, since `M114` will most likely not be an option anymore in case of catastrophic communication errors/power outages leading to a print failure, and anything else will be off and/or add too much processing overhead. 

And you also don't need to constantly write stuff to disk, you need to write it when stuff goes sideways (and if stuff goes that much sideways that the whole Pi crashes (side note: USB battery packs are great backups...) any kind of regularly created recovery point will be too much outdated for successful recovery anyhow).

SlightlyCyborg, 2017-05-03:
Also, I was ignoring the temp polls and was assuming user error if manual commands were sent, but ya, using byte position is the way to go.

When someone does want to recover a print, do you advise digging into _comm._currentFile and seeking to that position. I see now that the file has been abstracted away and _currentFile is PrintingGcodeFileInformation. So I could just grab and save _currentFile.pos

And then when someone wants to rescue their print, they will select the file, and then click a button in my plugin which fires this code

```
pos = open_bak_file_and_grab_byte_position()
self._printer._comm._currentFile.seek(pos)

#Since gcode is being started from the middle, the plugin should have default startup gcode that heats the printer, sets units, uses position absolute, etc. 
commands = get_warmup_commands_from_plugin_config()
self._printer.commands(commands)

self._printer.start_print()
```

Also, right now I am only considering using absolute coordinates since that is what my Sli3er setup uses. The only G92's it sends are for the extruder. I would home X and Y and send G92 at the beginning with X:0 Y:0 Z:<last known z position>

foosel, 2017-05-03:
Please explain to me why you desperately want to reinvent the wheel and not just make use of the recovery data that already gets saved, because right now you are basically requesting help to duplicate things (in a somewhat messy way) that are already there, working and available to plugins as proven in [the linked example/proof of concept](https://github.com/foosel/OctoPrint-PrintRecoveryPoc/blob/master/octoprint_printrecoverypoc/__init__.py).

> I would home X and Y and send G92 at the beginning with X:0 Y:0 Z:<last known z position>

What I forgot earlier: You also want the last known feedrate (`F`), because otherwise your firmware will use whatever it booted up with/last used instead on print resume, which might be way too slow or way to fast and mess up your print for good. And keep in mind that depending on absolute position and just having the next line from the gcode take care of your positioning will make you lose whole lines from your print.

SlightlyCyborg, 2017-05-03:
> Please explain to me why you desperately want to reinvent the wheel and not just make use of the recovery data that already gets saved, because right now you are basically requesting help to [...]

I didn't realize you already wrote a plugin for recovery until this comment. The file manager's recovery data was complex to me as I am not familiar with the code base. 

I also was not going to write anything more complex than what you wrote in your POC, so I am closing this issue since what I was planning on writing was already solved.

closed 2017-05-03 by SlightlyCyborg

foosel, 2017-05-03:
That plugin is only a proof of concept. It does allow resuming a print from a byte position, but else - like temperature tracking and such - is missing, as mentioned in its readme. It's not supposed to be used "in production" since it really is just a proof of concept, as reflected by its name, and lacks a lot of basic functionality a proper recovery plugin would need. I mentioned it way way [up there](https://github.com/foosel/OctoPrint/issues/1889#issuecomment-298904533) (as did @trgtylcnky) as a point of reference, NOT as a case of "already done". This definitely needs more work, and someone else to tackle it because I simply do not have the time next to maintaining OctoPrint itself, so if you or anyone else participating in this thread is interested in taking up refining it/writing something more sophisticated that would be heartily welcome (and I'd also reopen this ticket in that case if there is interest)!

Also sorry for the somewhat rough tone of my last comment. I was just very confused that you repeatedly appeared to ignore all mention of it over the last couple of hours and instead asked questions that would lead to basically reimplementing the core functionality that the POC depends on, and that confusion could be read differently than intended as I now realize.

trgtylcnky, 2017-05-18:
Hi,
What is the best way of restoring values like temperatures and active tool? Should I record them with the file position when print is interrupted or look into the GCode during recovering and get the latest values?

foosel, 2017-05-19:
The active tool is already monitored, temperatures are too (and that will become a bit more usable with 1.3.4, just didn't finish that in time for 1.3.3). But I guess it might sense to also persist that data as part of the print recovery file that's already created.

CameronLamont, 2018-01-21:
I had been wondering about this function myself. I am running octoprint on an old netbook which has a battery (have been looking at a simultaneous charging power bank for rpi).  My thinking is if the power went off the printer will go off but the battery would keep linux running. 

I don't know what octoprint does if the printer is pulled out during a print but in this case the 'printers brain' (arduino or the like) tends to stay on if usb is hooked up. My point being this would be a different state where motors are off but not the controller. Perhaps I will experiment here.  I will try out the poc too. 

An additional feature would be for the Linux box (rpi, netbook or laptop) to recognise it is now on battery power and save state persistantly. Sounds like using the existing 'wheel' here would be a good idea. 

In addition to emergency power outages I thought this might be useful if you wanted to relocate your printer mid print. (This might be a problem only I have lol)

Vexatuspl, 2018-03-01:
Anyone checked how it works with the MKS TFT? There are two options for it:
1. Without UPS: extruder stays where it was, after it warms up, the print continues.
2. With UPS: it only needs couple of minutes (or even less) to finish current command, move extruder away and store the progress.

I mean - it works  with Marlin! So why can't it be done on OctoPrint with the same approach? I've even seen description of the provoke protocol this TFT module is using...

foosel, 2018-03-05:
> it works with Marlin! So why can't it be done on OctoPrint with the same approach?

Because (any) firmware has more information about the printer's state than it provides on the serial interface that hosts rely on to do their work. In a nutshell: I'm trying to make the best I can here while being blind on one eye and deaf on both ears.

Vexatuspl, 2018-03-05:
You're doing a great job with OctoPrint. I'm just saying that the MKS TFT thing is more like a 3rd party module to the Marlin board then an integrated function - it's a host and it does handle the pause/resume functions. It even does support other firmwares the Marlin.
How I see it is that OctoPrint does know which line of gcode it has sent. If we would give a printer let's say a minute of backup power, it'll finish buffered gcodes. So (unless it's a belt driven z Axis printer) the nozzle will stay where it was. The only missing part is to restore temperatures. Our am I missing something?

foosel, 2018-03-05:
You are missing that once the steppers lose power, your position is lost as well and your nozzle can be pretty much anywhere (even a couple of microns of movement can ruin your print thanks to shifting). So just blindly assuming that the nozzle hasn't moved after a power loss is a recipe for failure. A rehome is necessary at least in X and Y (in the hopes that the accuracy of the home switches suffices). Ideally also in Z but that will probably not work because a half printed model is in the way, so you have to cross your heart and hope to die that the last Z you knew from before is still indeed the right one and tell the firmware "yo, you are at Z=..." through `G92`.

You are also missing the various buffers in between. OctoPrint knows what the last line is that it sent to the firmware. It also knows what the last line it read from the file it's supposed to print was. But it doesn't know if the last line it sent has actually made it from the serial line into the firmware's command buffer, or if maybe even made it into the planner buffer or actually got executed, if the user fiddled around with extrusion or flowrate multipliers or anything else through the printer controller, bypassing OctoPrint in the process. It doesn't know where the print head is (no, just adding up commands it sent to the printer does not work, see above, buffers). It doesn't know how fast it was moving at the moment of the catastrophic failure. 

For pause and resume OctoPrint tries to make do with some GCODE magic and educated guesses and you can utilize that through [custom pause and resume scripts](http://docs.octoprint.org/en/master/features/gcode_scripts.html#more-nifty-pause-and-resume).

But for catastrophic failure (= disconnect, power loss) with regularly available firmware interfaces its hands are pretty much bound. It already stores the last line it read from the file to print (in the hopes that made it through to the printer) and a bunch of other metadata when an error occurs during a print job or it gets cancelled *for plugins to experiment with print recovery*. But that is all it can do given the limited amount of information available to it.

Vexatuspl, 2018-03-05:
This is why I'm thinking about the option with some cheap ups that would provide enough power to at least complete buffered tasks and allow OctoPrint to store the progress. So it would be more like pause and resume. But OctoPrint would need to be able to restore after boot up. 
Do you think that's doable?

foosel, 2018-03-05:
Doable: yes. Something I'd put into core: no (next to no one has a UPS).

Vexatuspl, 2018-03-07:
Well, a good PSU capacitors last even for couple of seconds for the logic part at least. The smoothieware supports M601 which can be resumed by software. I'm just not sure if it completes all buffered gcodes before pausing?