OctoPrint/OctoPrint #1891: user management broken

trivalik, 2017-04-29:
I installed a fresh image of OctoPrint and update then to latest version. Now I am not able to create new user nor to change its password it the settings. The confirm button does nothing.

ntoff, 2017-04-30:
>I did not find in FAQ where it is: but good that I have to much time: /home/pi/.octoprint/logs/octoprint.log

Just for future reference there are links to all the logs in the settings menu (spanner icon)

foosel, 2017-05-02:
Thanks for reporting, this should now be fixed on the `maintenance` branch, soon `devel` and will go into the 1.3.3 release.

----

> I did not find in FAQ where it is: but good that I have to much time: /home/pi/.octoprint/logs/octoprint.log

Also for future reference: How to find everything that's requested from you in the ticket template is all explained in the [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md) as linked on top of the new issue form, repeatedly hinted at in the pre filled contents of the new issue form, very roughly summarized in the issue template itself and also linked to by the bot :)

foosel, 2017-05-31:
1.3.3 was just released.

closed 2017-05-31 by foosel