OctoPrint/OctoPrint #1893: Cura profile upload works even when no file is selected

eyal0, 2017-05-01:
#### What were you doing?

1. Go to Settings.
2. Select CuraEngine
3. Press Import Profile...
4. Browse for valid profile.
5. Press Confirm.  The profile will be uploaded
6. Press Import Profile... again.  This time, don't select a profile.
7. Press Confirm.  The profile will be uploaded again even though no profile is shown.

This is what the screen looks like right before step 7.  The Confirm button on this screen works: 
[image]

#### What did you expect to happen and what happened instead?

I expected the Confirm button not to work when there is no profile selected.

#### Branch & Commit or Version of OctoPrint

OctoPrint/devel
OctoPrint 1.3.2.post110+gd1d94624 (HEAD branch)

I have read the FAQ.

---

I'm working on the solution to this in https://github.com/OctoPrint/OctoPrint-Slic3r .  I'll solve it there and then port a similar fix for the cura plugin and send a PR.

eyal0, 2017-05-01:
This is the fix that I am porting to cura.js: https://github.com/OctoPrint/OctoPrint-Slic3r/commit/dbd31ed270af2fcfaed07bce117b66de4f18d57a

foosel, 2017-05-02:
I merged your PR manually via a cherry-pick, see the above commit. Additionally I did some further refactoring because I noticed that the metadata of an uploaded profile didn't update properly after selecting the file to upload. I also made it so the "Confirm" button is actually disabled until you select a file to upload, and swapped out your manual click binding on the file select element with a knockout data bind (which now is possible thanks to your earlier refactoring).

All in all I think that should be pretty solid now. I briefly looked into completely going away from using the `add` callback on the file upload element and instead going for only using `submit`, but I ran into some issues there. Needs another look at a later date.

Anyhow, this is already on `maintenance`, will soon be synced up to `devel` as well and be part of the 1.3.3 release.

Thanks!

commit f5fad518 (2017-05-02): Clear upload data when Aborting cura profile upload modal.

This fixes #1893.

Also, only bind click to the "Confirm" button once instead of binding
and unbinding it for each file.  Binding it just once seems cleaner.

(cherry picked from commit 83d1921)
closed 2017-05-31 by foosel