OctoPrint/OctoPrint #1896: Octoprint startup times....

rewolff, 2017-05-02:
I'm typing this while I restarted the octopi server because it was taking outrageously long for the octoprint server to start. It has consumed a full three minutes of CPU time, 

when the log reports: 
2017-05-02 08:25:30,669 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2017-05-02 08:25:32,133 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.235.20/'}
2017-05-02 08:25:39,715 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0-RC2 From Archive"
Then.. after 3:30 of CPU time, the server suddenly becomes responsive.

ntoff, 2017-05-02:
Do you perhaps have Avast antivirus? I recently discovered it caused octoprint to take forever to load, and or time out. I have no idea what Avast is doing back there but turning off the "web shield" fixed the issue.

rewolff, 2017-05-02:
No, I use Linux exclusively. No antivirus.

foosel, 2017-05-10:
Please provide a *full* log. This is not optional. Your excerpt doesn't tell me the whole story. Also provide your `.octoprint/data/preemptive_cache_config.yaml` file. Define what you mean with "responsive". Describe in detail what exactly you are seeing. You might think you already have, but sadly I'm not able to paint a clear picture from what you so far communicated.

rewolff, 2017-05-10:
```
edifix:~> cat .octoprint/data/preemptive_cache_config.yaml
/:
- _count: 102
  _timestamp: 1494140584.493292
  base_url: http://edi:5000/
  path: /
  query_string: l10n=en

```

[octoprint.log.2017-05-02.txt](https://github.com/foosel/OctoPrint/files/989470/octoprint.log.2017-05-02.txt)

With responsive I mean that I can access the GUI through my webbrowser.

foosel, 2017-05-10:
So "unresponsive" means "it doesn't load at all" or "it does load but takes a long time to finish loading"? Do you see the "OctoPrint is still starting up" screen and that takes long to switch over to the actual UI, do you see "OctoPrint started, refreshing page" and that takes long or do you see nothing while it's loading?

rewolff, 2017-05-10:
Ehh. I don't remember. Let me retry the experiment later on. I'll be back.

closed 2017-05-17 by GitIssueBot

ClusterM, 2018-06-26:
What is "Preemptively cache" at all? And why It's spending so much time for *every* used host, IP and port. I can wait 40 more seconds but loading time increasing after *every* IP change:

```
2018-06-26 02:27:58,729 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://10.13.0.101/'}
2018-06-26 02:29:07,515 - octoprint.server.util.sockjs - INFO - New connection from client: 10.13.0.1
2018-06-26 02:29:48,231 - octoprint.server.preemptive_cache - INFO - ... done in 109.50s
2018-06-26 02:29:48,239 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://10.13.0.51/'}
2018-06-26 02:30:27,858 - octoprint.server.preemptive_cache - INFO - ... done in 39.62s
2018-06-26 02:30:27,869 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://10.13.0.101:5000/'}
2018-06-26 02:31:08,371 - octoprint.server.preemptive_cache - INFO - ... done in 40.50s
2018-06-26 02:31:08,379 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://10.13.0.51:8080/'}
2018-06-26 02:31:48,162 - octoprint.server.preemptive_cache - INFO - ... done in 39.78s
2018-06-26 02:31:48,170 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://3dprinter/'}
2018-06-26 02:32:27,399 - octoprint.server.preemptive_cache - INFO - ... done in 39.23s
```

I will not change IP/port anymore but it's not good. Will it clean cache for old records automatically?
I deleted /home/octoprint/data/preemptive_cache_config.yaml and startup time is very fast now. 
But it's caching again on the next boot:
```
2018-06-26 03:01:09,694 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://10.13.0.51/'}
2018-06-26 03:01:34,674 - octoprint.server.util.sockjs - INFO - New connection from client: 10.13.0.1
2018-06-26 03:02:57,170 - octoprint.server.preemptive_cache - INFO - ... done in 107.48s
```
I can connect to Octoprint during caching but it's very slow since Octoprints loads CPU on 100%. So why I need this cache if it much faster without it?

Version: 1.3.9rc1
My preemptive_cache_config.yaml (after it was recreated):
```
/:
- _count: 1
  _timestamp: 1529973295.312376
  base_url: http://10.13.0.51/
  path: /
  query_string: l10n=en
```
Full octoprint.log after sturtup with it: https://pastebin.com/CWCYF5H4

I'm removing /home/octoprint/data/preemptive_cache_config.yaml using startup script now. Happy with it.

kolaha99, 2019-07-12:
> 

> I'm removing /home/octoprint/data/preemptive_cache_config.yaml using startup script now

how does you made it?