OctoPrint/OctoPrint #1897: Clicking on the test snapshot button shows multiple loading animations and windows

orc886, 2017-05-02:
OctoPrint 1.3.2 installed on Pi

Went to webcam at the settings window
Clicked on the "Test" button for the snapshot a few times and it added many loading animations to the button, and then opened 4 windows with the snapshot one over the other

[image]

orc886, 2017-05-02:
Updated

foosel, 2017-05-02:
Fixed as of the above commit:

[image]

Available on `maintenance`, soon `devel`, will be part of the upcoming 1.3.3 release. Thanks for reporting! 👍

foosel, 2017-05-31:
1.3.3 was just released.

closed 2017-05-31 by foosel