OctoPrint/OctoPrint #1900: Snipped using lessthan in if clause not working

ChinookTx, 2017-05-03:
#### What were you doing?

I am trying to create a snippet to be used in a gcode script using and if statement and the lessthan condition.  My code is as follows

{% if last_position.z is lessthan 100 %}

This is not super well documented anywhere that I can find so I'm assuming I can use it that way.  I would expect it return true or false based on the last_position.z.

#### What did you expect to happen and what happened instead?

In the logs, I get:
TemplateAssertionError: no test named 'lessthan'

The print hangs (despite being done) and the print head remains on the printed part.

#### Branch & Commit or Version of OctoPrint

Version 1.3.2 (master branch)

#### Operating System running OctoPrint

Running on OctoPi

#### Link to octoprint.log

Relevant log messages here:
https://pastebin.com/embed_js/keJ8r1d2

Salandora, 2017-05-03:
Could you try to use

{% if last_position.z <= 100 %}

I believe the one you tried is the jinja2.9 way and octoprint actually sticks to <2.9 which might explain why the lessthan function does not exist.

foosel, 2017-05-03:
What @Salandora said. Stick to Jinja 2.8 syntax for now. We can't upgrade to 2.9 since it has a change in behaviour that breaks backwards compatibility for plugins, see also #1697 and pallets/jinja#641.

However, your problem made me realise that for some reason there appears to be no 2.8.x version available of the Jinja documentation on the project's official page. I've therefore compiled said legacy documentation and made it available on [jinja.octoprint.org](http://jinja.octoprint.org) (if this still gives you an error that's because the DNS entry is too new, might take up to a day to resolve). OctoPrint's own documentation will be adjusted to point to that too with the next release.

foosel, 2017-05-03:
Documentation updated, to be rolled out with 1.3.3

ChinookTx, 2017-05-03:
Hmmm... I had also tried it with '<' yesterday and got the same (well similar) error in my logs:

TemplateSyntaxError: expected token 'name', got '<'

I'll try with <= just now to see if it makes a difference.

Pastebin here:  https://pastebin.com/embed_js/bsNv4kUE

foosel, 2017-05-03:
Try 

    if last_position.z < 100

NOT

    if last_position.z is < 100

That added `is` there is what's causing your syntax error, it doesn't make sense there. You can either have a comparison (`<=`) *or* a test (`is`), not both together. Also refer to the [template syntax documentation](http://jinja.octoprint.org/templates.html) (especially [the section about Comparison Expressions](http://jinja.octoprint.org/templates.html#comparisons)).

In general, if your script generates a syntax error, your script doesn't follow valid Jinja template syntax and there is nothing that OctoPrint can do anything about.

ChinookTx, 2017-05-03:
DUH!  That was it.  Print finished fine and my script ran as expected.  Thanks!

commit f1254622 (2017-05-03): Docs: Link to Jinja 2.8.1 documentation mirror at jinja.octoprint.org

The official docs are only 2.9+, that doesn't help us as long as we are
stuck with 2.8.

Fixes #1900
closed 2017-05-31 by foosel