OctoPrint/OctoPrint #1901: Octoprint causes Marlin 1.1.x to bootloop

discojon, 2017-05-04:
#### What were you doing?

Clean install of Marlin 1.1.x configured to work with Lulzbot Taz 5.  Printer works fine when not connected to Octoprint.  Once connected, the printer goes into a boot loop until disconnected.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Lulzbot Taz 5

#### Link to octoprint.log

https://pastebin.com/9igk6MQ3

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting any kind of communication issues between OctoPrint and your printer.
Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

Grogyan, 2017-05-04:
This a known bug with the Marlin 1.1.0 firmware, which has already been
patched today in the RCBugfix branch

discojon, 2017-05-04:
Thanks.  Which Commit fixed the issue?

Grogyan, 2017-05-04:
https://github.com/MarlinFirmware/Marlin/pull/6584

discojon, 2017-05-05:
Solved by using RCbugfix_new branch

closed 2017-05-05 by discojon