OctoPrint/OctoPrint #1902: [bug] "Saving" bar stays for ever after uploading files locally

snowdd1, 2017-05-05:
#### What were you doing?

Upload gcode files to OctoPrint.
Either using drag and drop, or the select-file dialog

#### What did you expect to happen and what happened instead?
I expect the animated bar saying "saving" to vanish after (a few) seconds, and I should be able to load or to print the uploaded file.
Instead, the animated bar stays forever, and the print and upload buttons next to the newly uploaded file are dysfunct - Until the web page is refreshed using F5 or reload.

#### Branch & Commit or Version of OctoPrint
`OctoPrint 1.3.2 (master branch)`

#### Operating System running OctoPrint
Raspbian started as OctoPi
`Linux octopi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux`

#### Printer model & used firmware incl. version
Anet A8-B V1

#### Browser and Version of Browser, Operating System running Browser
Reproduced with 
- Firefox 53.0 on Win 10
- MS Edge on Win 10

#### Link to octoprint.log

https://gist.github.com/snowdd1/60c559037ffe7df2a4c58871a59547b1

*192.168.178.135 is the windows client PC with Firefox 53.0*

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

https://gist.github.com/snowdd1/b233ec674258480cfbed18df7d96c61c

#### Screenshot(s) or video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2017-05-05:
Your JavaScript console hints at an error caused by one of your installed plugins causing this issue. Please try to reproduce this error when running in [safe mode](http://docs.octoprint.org/en/master/features/safemode.html) and report back.

snowdd1, 2017-05-05:
You were right. The "saving" bar vanishes almost immediately. Now trying to find the plugin that's causing that behavior.

snowdd1, 2017-05-05:
Checked all the plugins. It's the "Cost Plugin" https://github.com/jasiek/OctoPrint-Cost  
If disabled, everything is working fine.
Closing this, opening reference ticket at @jasiek 's repository

closed 2017-05-05 by snowdd1

derpicknicker1, 2017-07-12:
You used an old version of the plugin. Update to new version and the issue is gone (fixed jun 2016)