OctoPrint/OctoPrint #1903: i Found a 0day exploit on your System

rishikumar07, 2017-05-07:
Hi, i am rishikumar an security Researcher.

i found a 0day exploit on your systems.

i have mailed you the all details on contact@octoprint.org.

Reply to my mails.

Sincerely,
Rishikumar

rishikumar07, 2017-05-07:
hey this bug is not on the Raspberry Pi image OctoPi nor any unbundled third party OctoPrint plugins.

Thats a 0day exploit, if i posted in public then that bug is can exploited by anyone.

Just mail me. We can discuss about this privately.

foosel, 2017-05-08:
Neither a "bug" nor a "0-day exploit". To quote what I already posted on your [mailing list post](https://groups.google.com/d/msg/octoprint/vxnlIiaQGJo/LSbQA2yDCAAJ) on the matter:

> First of all - if you find an actual exploitable vulnerability in a piece of  software, you do NOT post this to its public mailing list, you contact the authors privately. That is called "responsible disclosure" and what security researchers usually do in order to prevent harm to come to users of affected systems. A sensationalist post like this one here is completely out of line and nothing I'd expect from an actual security researcher, especially since you take exactly NO measures in ensuring to keep the information on the "exploit" safe from potential attackers ("Just email me for details" is NOT proper vetting and completely irresponsible).
>
> You claimed that there are "thousands of vulnerable printers" that "are allowing anyone to access or change or print anything [on] the printer". OctoPrint on first setup prompts users to create a local user account in order to restrict access to anything that is not just monitoring capabilities. It strongly advises to create that account to prevent exactly the kind of access you are describing and discourages to disable access control in anything but completely controlled and strictly local networks. There are probably still users out there that ignore that advise and happily put their completely unsecured OctoPrint instance on the internet - that's sad, but nothing I can do anything about.
>
> So, in a nutshell, this "exploit" is people putting their OctoPrint instances online, potentially without the built-in access control, mostly without any additional means of security (e.g. basic auth), discoverable via some basic queries on shodan.io. Like millions of routers, NAS, IP cameras and other devices and software packages administrated by end users. I recommend against doing this, but it's your setup, not mine and hence your responsibility.

closed 2017-05-08 by foosel

rishikumar07, 2017-05-08:
i don't like you are saying about me.  you don't give respect to me. first i sent mail to them. but they don't reply to my mails. that why i am posted here. if you are respect me in my mail then why i am posting this here.
1st believe you are irresponsible. don't blame the Researchers. don't put your fault to others.

i think to keep the information private. that y i say to mail me. but you never sent mail to me. so who are the fault?

first believe you are don't have any security.

i definitely say you don't care about your security.

First give respect for the Security Researchers. this is the way to treat the researchers?.

i am placed on the Google security researchers list. you are saying that i am irresponsible?

foosel, 2017-05-08:
You sent your mail yesterday. I don't work on Sundays, I'm a human being, I need time off just like everyone else. I was in the process of replying to your mail first thing this morning. Then I saw your posts on the mailing list and this tracker. With these sensationalist posts you forced me to react publicly, which I did, outlining the issue and why it's outside of my control and also not the issue you make it out to be. If you take this as an offense against your person then I'm sorry about that, but I can't change how you perceive the statement of simple facts.

> don't blame the Researchers

I'm not. I'm merely pointing out how with an actual exploitable vulnerability your specific approach to communication ("I found an 0-day, anyone just mail me for details") would put people at risk.

Don't blame software authors for how end-users use their software and expect them to fix things that can't be fixed. I can't do more than tell people "don't put your OctoPrint instance on the internet without securing it". It doesn't do that by itself, and I can't prevent users from doing it. What you are doing here is the same as if you blamed and outright attacked a car manufacturer for people for getting their car stolen when they leave it unlocked and with the key in the ignition in some shady neighborhood.

> respect [...] respect [...] respect

Considering that your first reaction to a reply on the mailing list was to tell the other person "to go f- themselves", I'd say you shouldn't try to educate other people about respect.

And since this is apparently leading nowhere apart from attacks against my person and work ethic and least of all not to an understanding from your side why this was not the best approach in communication and the issue is not a software issue in the first place, I'm locking this now.