OctoPrint/OctoPrint #1905: [Request] Choices for time estimate

Snortimer, 2017-05-08:
I know that time estimates have been visited in the past many times, but I have a suggestion.

Can we have a choice in how the estimates are generated? In the settings, can there be a choice between two methods, with the understanding that one method might be a little more processor intensive?

One option could be the current method (that's HEAVILY influenced by slow first layers), but I had a thought about a second option:

Estimate how long the full print will take, then, every 5 minutes, or every layer up to 5 layers (and then every 10 layers or so), check the estimated time compared to the elapsed time, and adjust the overall estimate by the same percentage that it's off. I feel like that would likely give a very accurate result after only a few layers, and would be further refined as the print progressed.

Thanks for considering this!

-Joel

foosel, 2017-05-08:
I tried something pretty much like this about two years ago (sampling points across the file generated during initial analysis and live comparison with that during the print) and the results were horrible :/ To this day I'm not sure if I did something wrong in the implementation or not, but I spent a significant amount of time on trying to find an error and couldn't, the results were just bad.

I've been meaning to open up the time estimation to plugins, which comes pretty close to your request. That would allow people to experiment with alternative estimator implementations, and maybe also combine various results. I think there never was a ticket for that though.

Snortimer, 2017-05-08:
Thank you for the speedy response! I'm not good at coding, so I don't know if I could write a plugin for it, but I like the idea of people being able to.

Patschke, 2017-06-27:
I installed octoprint somewhat like four hours ago and am already annoyed of the bad time estimation. I might have a look at it and try to improve it. E.g. it should be way more accurate to take the % of the extruder position instead of % of completed instructions. This won't rise the processor load in any way worth to be mentioned. 

Not sure if it is (without much effort) possible to do so. I'll might have a look at it at the weekend.

foosel, 2018-05-17:
OctoPrint 1.3.9 will ship with two new hooks [`octoprint.filemanager.analysis.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-filemanager-analysis-factory) and [`octoprint.printer.estimation.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-printer-estimation-factory) that will allow plugins to override both the initial gcode analysis implementation as well as the live print time estimation. 

That should allow people to experiment with this stuff and find better but possibly more hardware/workflow/setup specific approaches that provide more accurate results than a stock option that has to *always* work can.

Knock yourselves out with this ;)

foosel, 2018-07-25:
1.3.9 has just been released.

closed 2018-07-25 by foosel