OctoPrint/OctoPrint #1908: OctoPrint hangs in endless loop when sending M-Codes containing an asterisk

deepthought-64, 2017-05-09:
#### What were you doing?

Sending M-Codes which contained an asterisk (*) via the Terminal Tab.

Example: `M117 *** Test ***`

#### What did you expect to happen and what happened instead?

The commands being executed by the printer (with the asterisks escaped if that is possible?) or OctoPrint rejecting commands containing an asterisk.

Alternatively OctoPrint could just drop the command/line if it was rejected by the printer for the same reason a number of times. 

What instead happened:

The printer-firmware seems to interpreted everything that followed the _first_ asterisk as the checksum which in that case obviously was not correct. 

When sending the 'bad' command _while printing_, OctoPrint would start hanging in an endless loop which included the printer reporting a checksum-error, instructing OctoPrint to resend the command in question, OctoPrint happily re-sending the 'bad' command and the loop restarts. 
This condition keeps on going until I restart OctoPrint

When sending the 'bad' command while OctoPrint _being idle_, the printer reports the checksum-error as above but the the communication freezes for half a minute (or so) during which it's not possible to control the printer. After some time OctoPrint reports a "Communication timeout while idle" and after that eventually all returns back to normal. 

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

Linux (Arch Linux)

#### Printer model & used firmware incl. version

Prusa i3 MK2S - FW 3.0.10

#### Browser and Version of Browser, Operating System running Browser

Chrome - Version 59.0.3071.29 (Official Build) beta (64-bit)

#### Link to octoprint.log

https://pastebin.com/6gn2yNzP

#### Link to contents of terminal tab or serial.log

Crashed during pring:
https://pastebin.com/Nfez5RbX

Crashed while idle:
https://pastebin.com/L9CX4ha3

#### Link to contents of Javascript console in the browser

    (Not applicable)

#### Screenshot(s) or video(s) showing the problem:

    (Not applicable)

I have read the FAQ.

foosel, 2017-05-09:
That's not OctoPrint, that's your firmware. `*` is a reserved character in the "protocol" between hosts like OctoPrint and printer firmware, it's used for signifying the start of the checksum (the `*15` stuff at the end). Since there's no (defined) way of escaping characters sent in e.g. `M117`, there's also no way to tell the firmware "hey, this is an actual asterisk, leave your hands off it. And apparently your firmware is looking for the first occurance of an `*` in a received line to extract the checksum, not the last, and that makes things of course fall on their face.

Sorry, but I can't fix this in OctoPrint. The only thing I could do is filter `*` out from sent GCODE lines, but that might make people angry whose firmware does handle this properly.

deepthought-64, 2017-05-09:
Hm, yeah, that was what I thought (and also wrote). 

But wouldn't it make sense to detect the endless loop and do something about it?
When all the previous commands were all transferred successfully and suddenly the printer requests resending of the same line over and over again something must be fishy...!?

foosel, 2017-05-09:
Sorry, I was doing too many things at once again 😞 

The thing is, there are a lot of things that could be fishy. Firmware requesting the same line over and over again can be noise on the line, a brown-out or some other kind of bug. And trying to be clever in such cases will just lead to hard to debug issues when that cleverness leads to wrong behaviour.

As a general rule of thumb, OctoPrint keeps its hands off of what it reads from the printed files or receives from the user as typed in via the terminal. There are some very carefully chosen exceptions to that rule (e.g. rewriting an `M110` to always include a line number, or applying temperature offsets to `M104`/`M109`/`M140`/`M190`, or preventing heat bed commands when no heated bed is configured) and third party plugins might also rewrite things, but in general, OctoPrint pumps through GCODE exactly as it gets it from wherever. The slicer/user/whoever creates the code will in most cases know better why they did it that way than OctoPrint, especially in case of people playing around with their firmware, or less mainstream versions.

And if I introduced an exception to that rule here, what should OctoPrint do about this? It sees a line that it can't send because the printer rejects it again and again. This line could be anything (here it is an `M117`, but it might also be some other command that OctoPrint doesn't know if it's required for the print or not). So what is it supposed to do? Just skip the line? Bad idea, could be important. 

So aborting the print would be logical, but wouldn't help you here either. And the question is - how many repeats justify that, 5, 10, 50?

deepthought-64, 2017-05-09:
No worries. :)

From a software-engineering standpoint I completely understand your points. Especially when you say that whoever created the GCODE is responsible that it works correctly with the firmware of the printer.
The longer I think about it the more it occurs to me that this case should entirely be solved on the printer-firmware's side. 

Nevertheless I found that odd behaviour and thought it might be worth reporting. :)

Schöne Grüße aus Wien :)

foosel, 2017-05-09:
Yeah, the clean solution here would actually be that the firmware doesn't look for the first occurrence of the `*` but the last one instead. Everything would just work fine in that case.

Definitely good to know though that firmware does that. Might make sense to file a bug report with Prusa Research and/or the official Marlin. From the looks of it, they have the same issue:

     char* apos = strchr(command, '*');

Should rather be a `strrchr`...

Will you do that? Otherwise I'll try to get around to it myself, stuff like that needs to be reported.

If you don't mind though, I'll close this issue here now, since there's nothing I could make OctoPrint do to work around this which wouldn't introduce issues in some other situation. Thinking about maybe looking into a "endless resend of the same line" abort though - will open a new ticket for that after some more thinking about what that might break.

closed 2017-05-09 by foosel

deepthought-64, 2017-05-09:
I just reported this issue here: https://github.com/MarlinFirmware/Marlin/issues/6655

fiveangle, 2017-05-09:
PR doesn't typically track upstream so if you want it fixed for your printer you should probably file a defect with PR

deepthought-64, 2017-05-10:
@fiveangle Thanks, good to know. 
But nevertheless this could also be interesting for the Marlin-firmware.