OctoPrint/OctoPrint #1909: Update to 1.3.2 Fails

TechJeeper, 2017-05-09:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete any lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to repor tis still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the contribution
guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

When I click "Update" on the toast message in Octoprint, it begins the update then fails.  See logs.

#### What did you expect to happen and what happened instead?

#### Branch & Commit or Version of OctoPrint

1.2.10

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Duplicator I3

#### Browser and Version of Browser, Operating System running Browser

Chrome

#### Link to octoprint.log

https://pastebin.com/cFLHtFgy

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting any kind of communication issues between OctoPrint and your printer.
Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) or video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2017-05-10:
This is a freshly flashed OctoPi? It looks like something is seriously wrong with the checkout of OctoPrint on that one, making the update fail in a way I've never seen before. I tested updating fresh OctoPi 0.13 with 1.2.10 to 1.3.2 repeatedly during release and I never saw anything like this, so it must be something wrong with your specific installation.

It looks like something in the git got corrupted for whatever reason. You might be able to fix this by SSH-ing into your Pi and doing this

    cd ~/OctoPrint
    git fsck 2>&1 | grep "bad ref" | rev | cut -d " " -f1 | rev | while read ref; do echo "Deleting broken ref $(git rev-parse --git-dir)/$ref" && rm "$(git rev-parse --git-dir)/$ref"; done

The latter part is one line. Copy and paste it and make sure it comes out as one line, otherwise it won't work. What that does is try to find all corrupted references in your git checkout and delete them. The next attempt to update should then hopefully work (and recreate the references as required). Please report back if that works out for you.

But if that doesn't fix it, I can't help you either. If this is a fresh install, I'd suggest to try reflashing the image - something might have gone wrong on initial flash. Also make sure to never just pull the power on the Pi, but to always shut it down properly first, or file corruption may occur that might also lead to issues exactly like this.

In any case, this isn't an issue within OctoPrint itself but basically with its runtime environment in your local setup, which got corrupted for some reason. For questions like these, the [mailing list](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741) are the better location.

foosel, 2017-05-18:
@TechJeeper any update on this?

foosel, 2017-06-02:
Apparently not, closing.

closed 2017-06-02 by foosel