OctoPrint/OctoPrint #1910: [Request] Add extruder temperature to PrintPaused variables

TachyonStream, 2017-05-09:
When using the pause/resume scripts, it would be very helpful to have the last commanded extruder  temperature available, similar to the pause_position variables that were added.  This would allow the scripts to be used for filament change without needing to manually manipulate the extruder temperature.  May be useful for some to have the bed temperature available as well.  

The user could define the change-out temperature in the pause script, and use a command to resume the original temperature in the resume script.

Obviously this may have the same limitations related to multiple extruders as the current position values.

#### What were you doing?
Using Pause / Resume scripts for filament change.

#### What did you expect to happen and what happened instead?
Have command available to resume original extruder temperature.  Currently feature does not exist - must do manually or hard code.

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
MonoPrice Maker Select Plus (Wanhao i3 Plus)

foosel, 2017-05-10:
I have to admit, I'm a bit confused. Do you shut down your heaters during a filament change? Because otherwise I don't understand why you'd need to reset them.

But I agree that the latest temperature data would be interesting to have available in the GCODE scripts.

TachyonStream, 2017-05-10:
I turn my heaters down (not off) during a filament change (about 80-100c for PLA).  This has a few benefits.

1. Clean filament changes - the material is solid enough to pull out as 1 piece (no residual left in melt chamber), but elastic enough to still pull free.  Cleans the nozzle well too.
2. Prevents molten material from potentially getting into extruder gears, cold end, etc. 
3. Prevents heat creep from melting material in the cold end. 
4. It won't burn filament sitting in the hot end for an extended period of time (if you're not constantly at the printer).   Also will not ooze plastic. 
5. Very little purge needed with the new color, as the "cold" pull of material mostly empties the nozzle. 

Once I pull the old filament out, then I heat back up and load/prime the new stuff.

JohnGJ, 2017-06-11:
I have to say, I'd love to see this; I occasionally have to pause my prints for a period of time, so I turn off the bed and hotness. This means I have to hard code the temperature into the resume script, which requires that I have to keep changing the scripts whenever I switch materials (and it's easy to forget to do this!).

foosel, 2017-06-22:
Just pushed to `maintenance`, soon `devel`, will be part of 1.3.5

stuartsmi, 2017-07-30:
Would it be possible to include bed temperature in this change? Quite often, my filament will run out when I am not at home and it may be hours before I return. I have Octopi set to shutdown both heaters when filament runs out. I would like to be able to restore both heaters programatically rather than having to manually reheat after filament replacement.

foosel, 2017-08-21:
@stuartsmi the bed is of course already part of what was implemented in 263c297

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel

bruce356, 2018-09-04:
@foosel or others Hi, would anyone have a sample script to cool the hotend for filament change (to say 110deg) when hitting Pause button. 
On resume then increase temperature to the last temperature before the pause, once the hotend reaches temperature resume the print.
Thanks & regards - bruce