OctoPrint/OctoPrint #1911: Usernames and passwords sent as GET parameters.

theorician, 2017-05-11:
#### What were you doing?

Log in to the Octoprint interface using the top right Login button.

#### What did you expect to happen and what happened instead?

Expected: Log in form sends as POST
Actual: Log in form sends as GET

#### Branch & Commit or Version of OctoPrint

1.3.2

#### Operating System running OctoPrint

OctoPi

foosel, 2017-05-11:
Uhm... no?

[image]

And also:

```
$ curl -H "X-Api-Key: test1234" "http://localhost:5000/api/login"
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>405 Method Not Allowed</title>
<h1>Method Not Allowed</h1>
<p>The method GET is not allowed for the requested URL.</p>
```

Login isn't even possible via GET.

BillyBlaze, 2017-05-11:
It sounds like a javascript error happened and the form got posted as GET due to the javascript not able to call preventDefault on the <form> tag.

Perhaps a fault in a 3rd party plugin?

foosel, 2017-05-11:
Ah, that would make sense. It will probably remain a mystery though unless @theorician provides all information requested in the ticket form.

trivalik, 2017-05-12:
I saw this error also, maybe related somehow to the error with the password that äüö are not possible.

theorician, 2017-05-15:
@foosel I am out of town right now but I will try and reproduce and provide more info when I get back, soon. Thanks!

foosel, 2017-06-29:
Any update on this?

foosel, 2017-07-24:
Still no update on this. Closing for now since not reproducible and sadly also no additional info from OP.

closed 2017-07-24 by foosel