OctoPrint/OctoPrint #1912: get serial log when raspberry is getting offline

trivalik, 2017-05-12:
#### What were you doing?

In case you must switch of the printer, which is on same switch able power socket. In case of trouble you turn of that power switch, than is it without enabled serial.log not possible to get the logging.

#### What did you expect to happen and what happened instead?

Maybe it would be good, when the reconnect window appear, that there is somehow a possiblity to grab the log, Just save would be enough. But I guess for javascript is not easy to save, so view and user can copy manually is also enough.

But maybe is it more easy if the window which say you should reconnect, is possible to close with a X, andthen all tabs can be still viewed. The log and the temperature graph. So I would suggest the second.
Interfaces which forbid any analyse after a failure is sometimes wanted, but here no mechanical defect can follow. 

#### Branch & Commit or Version of OctoPrint

1.32

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

not relevant

#### Browser and Version of Browser, Operating System running Browser

Google Chrome 58

#### Link to octoprint.log

description for a feature, no log available

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting any kind of communication issues between OctoPrint and your printer.
Never truncate.

**serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]**

This bold text shows, the problem, if you have trouble and has a power lost, not log is then there.

#### Link to contents of Javascript console in the browser

Not sure where to get javascript console output, could not find in guide.

#### Screenshot(s) or video(s) showing the problem:

[image]

I have read the FAQ.

ntoff, 2017-05-13:
I believe what you're asking is for the ability to close the "server is offline" popup and overlay, so you can read / copy the remaining lines in the serial log, correct? If so, that is a feature I too would have been able to make use of in the past.

trivalik, 2017-05-13:
Yes, this is the description with good english.