OctoPrint/OctoPrint #1914: [RC Feedback] Feedback on the 1.3.3rc1 Release Candidate

foosel, 2017-05-13:
Please provide general feedback on your experience with the 1.3.3rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

ctgreybeard, 2017-05-13:
The print time is now cleared of the progress information when the print ends. I was copying the State information into my print log, especially the elapsed time, for my records.

ntoff, 2017-05-14:
on the timelapse page, everything gets greyed out except the "save as default" checkbox which you can still check and uncheck (although you can't save anything obviously).

trying to upload to SD while printing results in a big red error box, not sure if that box would appear after a lengthy period of trying to upload (I don't have an sd card so running on the virtual printer with a fake sd card so nearly everything happens instantly), would it not make more sense to disable the upload to sd button while printing? Unless it already does that when using an actual printer and not the virtual one?

Also there's the same error message for trying to upload a file and overwrite it while it's being printed (warns about it not being the right file type), and trying to upload to SD while printing. 

same looking error messages (top is the SD upload while printing error, bottom is upload overwrite error):
[image]

Only after clicking "more" is the actual reason revealed (maybe expand "more" by default?)

foosel, 2017-05-15:
@ctgreybeard good catch. Not sure yet I'll push out a new RC just for that, but it's fixed on the `maintenance` branch at least. See also #1917. 

@ntoff also good catch! Regarding the "save as default" checkbox: I'll fix that on `maintenance` as well, but not sure about putting out a new RC just for that and #1917. Still thinking about it.

About the SD upload thing: Definitely something I'll look into, but no regression (= "worked in 1.3.2, doesn't work in 1.3.3rc1") so that will have to wait until 1.3.4. Same goes for the "restart now" button - also something to fix, but no regression as far as I can tell.

Let's see what else will get reported back :)

foosel, 2017-05-16:
Ok, it's settled, I'll do a second release candidate tomorrow (not enough time left today for getting through the release checklist, got an appointment in an hour), fixing #1917, #1918 and #1919 (which were actual regressions).

@ntoff the other things you reported :+1: I have fixed on a local branch, ready to be merged into `maintenance` once 1.3.3 stable is out of the door, so that stuff will be tackled in 1.3.4.

ashpool, 2017-05-17:
`octoprint` did not restart after ugrade, had to ssh and `sudo service octoprint restart`

foosel, 2017-05-17:
@ashpool and what did it do instead? Also: is there a log? ;)

foosel, 2017-05-17:
Marking as closed because 1.3.3rc2 has been [released](http://octoprint.org/blog/2017/05/17/new-release-candidate-1.3.3rc2/). Comments are still possible though.

closed 2017-05-17 by foosel

ashpool, 2017-05-18:
I think there was an issue with my Pi. Everything worked fine after a (Windows style) restart 😄