OctoPrint/OctoPrint #1916: Webcam doesn't load if first tab is set to 'Control'

aviatorken89, 2017-05-15:
**What were you doing?**

Setting a custom appearance tab order so that the Controls is the first tab loaded in OctoPrint.

My custom order in the config.yaml file is:

`order:
      tab:
      - control
      - gcodeviewer
      - temperature
      - terminal
      - timelapse`

If the webcam is set to be the first tab, the webcam doesn't load initally until you manually click another tab and come back, or go to another tab in your browser and then come back to the OctoPrint tab.

**What did you expect to happen and what happened instead?**

I fully expect the webcam to load if it's set to your first tab. To get the webcam to load, you have to manually go to another tab in OctoPrint and then click on the 'Control' tab again, or simply open a new tab in your browser and then come back.

**Branch & Commit or Version of OctoPrint**

Devel and master branches

**Operating System running OctoPrint**

OctoPi

**Browser and Version of Browser, Operating System running Browser**

Safari 10.1 on macOS 10.12.4, and Firefox 53.0.2 on macOS 10.12.4.

I have read the FAQ.

I have fixed this bug easily here and I hope you put it into your next commit: [https://github.com/aviatorken89/OctoPrint/commit/d66a4f3cfe8fe7e9a3f855e2433df97f3aefe579](url)

foosel, 2017-05-15:
@aviatorken89 your fix sadly breaks the "do not load the webcam if it's not visible" behaviour for all other cases, so it's not the best solution to this particular issue. I'll see what else I can do. It will probably not make it into 1.3.3 though - I already put out the first RC for that and past experience has shown that it's best to limit changes after that to fixes for regressions only.

aviatorken89, 2017-05-15:
When I tested the fix, the webcam was still being disabled when I switched tabs, so I'm not sure how it would break anything?

tobiebooth, 2017-05-16:
I think the part that @foosel is missing is that the first line in the enableWebcam method actually short circuits if the tab is not "#control" which prevents this change from breaking other behaviors (as far as I can see).

foosel, 2017-05-16:
Indeed I did. Sometimes my own foresight surprises me.

foosel, 2017-05-16:
@aviatorken89 I've cherry-picked your commit to a local branch and will merge it (plus some other bits) to `maintenance` for 1.3.4 once I have 1.3.3 stable out of the door.

Sorry for the confusion, as hinted at this morning I indeed had simply forgotten about the follow-up check by `_enableWebcam` itself to ensure the stream is only enabled if the tab indeed is focused.

thedoble, 2017-07-12:
Hi Foosel, 
I just came across this bug myself when upgrading to 1.3.4. Previously I didn't have this problem but now I get the 'webcam stream loading...' message until I switch tabs off, and back to, the control tab.
Please forgive me if I am misunderstanding but should this be fixed in 1.3.4 or is it scheduled for a future release?
Thanks

foosel, 2017-07-12:
It was scheduled to go out with the next stable release after 1.3.3 which back then I thought would be 1.3.4, but then I had to put out a hotfix for a somewhat nasty issue a handful people were running into which got that version. So this fix won't be released until 1.3.5 goes out the door. See also the current milestone this ticket is in (and it not being closed yet ;)).

thedoble, 2017-07-12:
No worries, thanks for your reply!

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel