OctoPrint/OctoPrint #1918: 1.3.3rc1: "Save as default" on timelapse config not disabled alongside everything else when non operational

foosel, 2017-05-15:
Originally reported by @ntoff in #1914 ([see this comment](https://github.com/foosel/OctoPrint/issues/1914#issuecomment-301292361)).

#### Expected

"Save as default" is disabled when all other settings are disabled as well due to the printer being offline.

#### Actual

It is not disabled and can be toggled. Doesn't do anything though.

[image]

#### Solution

Fix knockout data bindings.

#### Sidenote

It would probably make sense to allow timelapse configuration when the printer is not operational...

closed 2017-05-31 by foosel

jkyoutsey, 2019-03-18:
I can configure timelapse, but I cannot check the "Save as Default" button.  I'm on 1.3.10

foosel, 2019-03-18:
@jkyoutsey can you please open a *full* new ticket for that? This one is nearly two years old and there have since been so many changes to OctoPrint that it's probably unrelated.

Gunnl, 2019-05-10:
@jkyoutsey , have you opened a new ticket? I am having same issue ...

jordanwjolly, 2019-09-24:
@jkyoutsey  I having the same issue

commit 6d73e655 (2017-05-15): Timelapse: Disable "save as default" along other controls

Fixes #1918