OctoPrint/OctoPrint #1921: [RC Feedback] Feedback on the 1.3.3rc2 Release Candidate

foosel, 2017-05-17:
Please provide *general* feedback on your experience with the 1.3.3rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

dookie2000ca, 2017-05-17:
Upgrade from rc1 to rc2 went smoothly on my OctoPi 0.14rc1 installation. Printer is an UltiBots D300VS running RepRapFirmware for Duet WiFi 1.15e.

MartinTMV, 2017-05-17:
Just upgraded my RaPi3, working without problems with my Wanhao D6 (Marlin)

tophstar, 2017-05-17:
Just I uni jioji9 hi k9 u

foosel, 2017-05-18:
@tophstar ?

ptrav, 2017-05-18:
Upgrade from OctoPrint-1.3.2 to 1.3.3rc2 went smooth with no issues to report. Printer is a Monoprice MP Select Mini, OctoPi distro, running on a RasPi 2 B+. No plugins or additional features enabled other than the main server.

foosel, 2017-05-24:
Marking as closed because 1.3.3rc3 has been [released](http://octoprint.org/blog/2017/05/24/new-release-candidate-1.3.3rc3/). Comments are still possible though.

New feedback ticket is #1935

closed 2017-05-24 by foosel