OctoPrint/OctoPrint #1922: [Brainstorming] Make OctoPrint more relocatable in the filesystem/compatible with distro packages

Depaulicious, 2017-05-17:
A big thing that always left me unsatisfied about OctoPrint is the way it is placed in the filesystem. It's very Raspberry Pi centric, and by that I mean that it's usually acceptable to put things in `pi`'s home in Raspbian, then run them as `pi`, but that looks very unclean to me anywhere else.

This _issue_ is more evident if you want to autostart OctoPrint on boot. There are sysv startup scripts, etc., and what they do is simply start it as `pi`, with the path _hardcoded_ to `/home/pi/...`.

I believe OctoPrint should be made more relocatable and, by default, placed in a system directory (`/usr/share`, `/opt`?), not only to make it "more elegant", but to better automate tasks such as setting up bootup scripts/units, supporting stuff like systemd's socket activation, and distributing it as distro packages.

### About distro packages
Though I really like the fact that OctoPrint is set up in a virtualenv, so that it doesn't mess up with the rest of the system, I feel like it's not the most convenient way to do it. I think OctoPrint could be modified a bit to support a distro package installation, in addition to the current virtualenv setup.

I know this is not a trivial task, which is why I'm writing here; I hope this will start a useful discussion.

I haven't thought very much about it, but this is a basic list of things that I think need to be done in order to implement this:

- OctoPrint should be able to install itself as a system Python module (I think that's already feasible) and detect that, so that it won't try to update itself and let the distribution package manager handle this part.
- It should also be able to install the plugins into some user-writable directory (`~/.local/share`?), so that 1) users can have their own independent OctoPrint daemon (don't really see the use case but why not :rofl:) and 2) it does not mess up with the system packages 3) It would be compatible with packaging systems like `snap`, which totally forbid the modification of the software image
- It should be started by startup scripts as its own unprivileged system user (`octoprint`?), like how every other daemon does.
- Startup scripts/units should be generated at "build" time, customized for the targeted packaging system

### Possible use cases

Other than seeing a nice `octoprint` package in popular distributions' repositories, the main reason why I think this is a thing to do is `snap` compatibility.

Snap combines the trouble-safety of a virtualenv with the ease of use of a distribution package, allowing users to simply `snap install octoprint` and be ready to go. It will be quite a pain though, as it's very restrictive on what program has access to which component of the system, but I think the result will be worth it.

I can already imagine a "smart IoT printer" for sale with OctoPrint preinstalled as a snap app on Ubuntu Core or similar, allowing users to simply reset the system to a known state with one button.

Let me know what you think about this ;)

foosel, 2017-05-18:
Well, the provided initscript is simply there for convenience in case you want to set things up in the most common way things are set up, and it might even make sense to move this to its own repository.

Everything else in OctoPrint's code base is completely platform agnostic, as in, no references to `pi` or anything else system specific for that matter, so I'm not entirely sure what you are referring to here. Could you provide some examples that are not actually OctoPi or "recommended setup on a Pi" related? 

The only thing that comes to mind is the default location for settings and such being `~/.octoprint` for the user OctoPrint is running under, but that could also easily be changed via an adjusted startup call.

> * OctoPrint should be able to install itself as a system Python module (I think that's already feasible) and detect that, so that it won't try to update itself and let the distribution package manager handle this part.

While the setup guide for Pi on the wiki instructs people to just throw it into the user home in its own virtual environment, that's simply for reasons of simplicity and also  doesn't hinder people from installing it in any other way (or on any other OS, e.g. Windows, Mac, FreeBSD, ...). In the end, OctoPrint is just a python package that you could `pip install` anywhere and in any form that you want. It does depend on specific versions of some package, for the one or other reasons, and this is something I do not want not change - I've run into too many problems with outdated python libraries provided by the system package manager that produced horribly hard to track down bugs.

> * It should also be able to install the plugins into some user-writable directory (~/.local/share?), so that 1) users can have their own independent OctoPrint daemon (don't really see the use case but why not 🤣) and 2) it does not mess up with the system packages 3) It would be compatible with packaging systems like snap, which totally forbid the modification of the software image

OctoPrint's `pip` helper which the bundled Plugin Manager plugin uses for plugin installation will also happily install into the [user site dir](https://www.python.org/dev/peps/pep-0370/) when it [detects](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/pip.py#L178-L199) that OctoPrint is installed globally and not running from a virtualenv (the user dir and virtual environments don't work well together according to me tests, hence that limitation). 

The bundled Software Update plugin doesn't ship in a way that it will attempt to update itself - in fact, the only reason why that works out of the box on OctoPi is because the necessary configuration to make that work is done on that image through `config.yaml`. If you'd install a stock OctoPrint somewhere in your system, you'd get a notification if a new version was there, but not be able to install it from within OctoPrint, it would be your job to somehow update to that new version. It would be completely possible to just do a similar configuration to fetch update information from `apt` or whatever package manager of choice you are thinking of and installing/updating OctoPrint through that instead. That could even be done via a configuration overlay done from a package specific plugin installed with the application.

> * It should be started by startup scripts as its own unprivileged system user (octoprint?), like how every other daemon does.
> * Startup scripts/units should be generated at "build" time, customized for the targeted packaging system

That's something that should probably reside in its own distribution/target platform specific package, not in the platform agnostic core repo (as I mentioned above, the current contents of the `scripts` repo violate that, I am aware of that, but that's also mostly for historic reasons and reasons of simplicity, I'd be happy to move that off someplace else). 

What IS in fact lacking - but what I don't see as something that should be part of this core repository but instead of a separate one - is proper packaging for e.g. Debian. I have absolutely no experience with that kind of stuff and also have my hands completely full already with everything else. I always hoped that if there is demand, someone would step up and help making this a reality, but so far nothing has happened ;)

The open TODOs that I currently see are:

  - finding someone who'd take care of distribution specific packaging
  - possibly finding a way to realize support for the release channels (because that currently relies on the `git` based update mechanism, simply because of cost vs. usefulness for a more sophisticated solution)

But maybe I'm missing something completely obvious here, if so, please tell me.

Depaulicious, 2017-05-18:
> The open TODOs that I currently see are:
> - finding someone who'd take care of distribution specific packaging

I can take care of that; I don't have much time so it might take a while.

> - possibly finding a way to realize support for the release channels (because that currently relies on the git based update mechanism, simply because of cost vs. usefulness for a more sophisticated solution)

Well with distro packages that would be very distro-specific. Only thing we can do is somehow tell OctoPrint how it's been installed and add code to the software update plugin to handle the most basic cases.

I'll try to set up debian and arch packages first and put them on launchpad/AUR, with automatic startup script setup and all the basic stuff. Then eventually I'll submit code to ask the package manager if updates are available.

foosel, 2017-05-19:
Sounds good!

The thing that I keep thinking about off and on again (#showerthoughts) with regards to distro packaging is how to handle updates. Usually I'd say, a userspace application shouldn't be allowed to update itself if its installed globally (and permissions should be set accordingly to prevent this). However I could imagine that causing issues for a lot of OctoPrint's less technically inclined users that prefer being able to press a simple button. A general addition of e.g. "apt-get" to sudoers sounds like a bad idea, but a specific update script (write protected against the `octoprint` user) that simply does "sudo apt-get update && sudo apt-get upgrade octoprint" or equivalent might work in such cases.

> add code to the software update plugin to handle the most basic cases

To a certain degree that's already possible, since the plugin allows defining a specific python function for checking and also executing updates. However, that's hard to utilize for OctoPrint's own updating in particular (since that you configure via `config.yaml`, not through actual code), so I guess it's time to look at allowing to define additional update check and execution methods through plugin hooks. What I do want to avoid is having `apt`, `pacman` and `yourFavouritePackageManager` specific code within OctoPrint, even if it's hidden inside the software update plugin. `pip` is cross platform, the others aren't. But with plugin hooks in place that should be a piece of cake to realize through a tiny platform specific plugin that would simply be installed by the package alongside OctoPrint itself.

Depaulicious, 2017-05-19:
We should take a look at packagekit, it should provide bindings to most package managers through policykit, and that means we can add pk rules to update OctoPrint without password.
I don't think, however, this is the most important thing to have. We could momentarily say in software update that "OctoPrint has been installed as a distro package and can't be updated automatically, check the documentation". Then add code to do things properly.

Depaulicious, 2017-05-20:
@foosel I just noticed this may be an issue for Debian packaging: https://github.com/foosel/OctoPrint/blob/master/THIRDPARTYLICENSES.md

Debian requires _every single one of them_ accurately listed in a machine-readable `copyright` file, including all the copyright dates etc.

I'm also not sure if they'd be happy with just your name as the copyright owner, as you also have [this](https://github.com/foosel/OctoPrint/blob/master/AUTHORS.md) in which you list a bunch of people. And it forked off Cura anyway, right?

I'll ask a friend of mine who's a Debian Developer and Ubuntu MOTU if that's a big issue.

foosel, 2017-05-30:
> Of course none of these are actual issues to distribute packages from unofficial repositories, but I think it's better to do things correctly in an open and transparent way from the beginning.

To be honest, I would first and foremost try to target the creation of a `deb` file, THEN worry about the intricacies of individual distributions, especially ones that are as strict as Debian (which based on past discussions I had with Debian maintainers is somewhat of a nightmare to cater to). I'd rather be able to offer up packages on an unofficial repository than none at all.

Depaulicious, 2017-05-31:
Well that doesn't seem like a huge issue to me. As I told you I have a friend that's a Debian Developer, and he told me that when I get a package to build we can meet and fix up everything else.

bzed, 2017-10-08:
I'll happily to review and sponsor package uploads. Just give me something to review :)