OctoPrint/OctoPrint #1923: SSH login no longer possible?

joeynuggetz, 2017-05-18:
This has happened twice. Both times running the pi image written to SD card. Randomly and with no warning, the pi will not take the default login on pi/raspberry. I never ever change that password and its now happened again. SSH into octo and I get an access denied message when entering the password. Is this a known issue? Has anyone else ever run into this? Tried resetting password last time it happened and couldn't do it. Is there a hack that is known to work so I can log in and see what's going on?

ntoff, 2017-05-18:
First of all, you're in the wrong bug tracker (this is for octoPRINT, octoPI is over here: https://github.com/guysoft/OctoPi/)

Second of all:

>will not take the default login on pi/raspberry. I never ever change that password

You absolutely should change that default password. My guess is someone has gained unauthorised access because you've left the password as default, and changed your password for you. If that is the case, be thankful that's all they did (and it may not be all they've done).

foosel, 2017-05-18:
As already pointed out, this is the wrong tracker. OctoPrint doesn't have SSH. OctoPi does, but I don't think this would be suited for that bug tracker either. That sounds more like a support request and for that the bot already referred to the Mailinglist or the G+ community, which are better suited for this than bug trackers.

Closing this now.

closed 2017-05-18 by foosel