OctoPrint/OctoPrint #1924: 404 error - Failed to load sockjs.js.map

aviatorken89, 2017-05-18:
**What were you doing?**

Viewing the javascript console in Safari

**What did you expect to happen and what happened instead?**

I expect there to not be a 404 error, but there was.

**Branch & Commit or Version of OctoPrint**

Devel

**Operating System running OctoPrint**

OctoPi

**Browser and Version of Browser, Operating System running Browser**

Safari 10.1 on macOS 10.12.4, and Firefox 53.0.2 on macOS 10.12.4.

I have read the FAQ.
[image]

foosel, 2017-05-19:
I couldn't reproduce this (for some reason neither Chrome nor Firefox did want to show me any information regarding referenced source maps loading or not loading), but I fixed it anyhow, even though it's merely "a bit ugly" but not really a problem - source maps are a debugging tool and not relevant for the functionality of the UI. 

Since source map resolution is wonky in combination with assets bundled through webassets, OctoPrint now strips source map references from bundled JS files. Additionally though I also added the source map for sock.js to the sources, just in case someone includes that js file directly (e.g. from a custom UI).

Relevant commits are 832c27ed18bd84b517a2b6de824cd9a96995ce7d and cfde6342a824712169cc7b6f9170c6e536b2a508 which should be available on `maintenance` and `devel`.

A quick report on whether that indeed solves this for you would be nice, since I couldn't trigger such an error in the browser console myself.

ntoff, 2017-05-26:
>I couldn't reproduce this (for some reason neither Chrome nor Firefox did want to show me any information regarding referenced source maps loading or not loading)

I see nothing but 404's in what I guess is stdout? the terminal windows thing when you run octoprint from a command prompt (what I assume ends up in octoprint.log). Not so much in chrome's console though.

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel